prusa3d/Prusa-Firmware-Buddy #4251: Filament change park location

chschmit, 2024-10-13:
### Printer model

MK4S

### Firmware version

6.2.0-alpha2+8702

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

When a filament change occurs the printhead now parks on the left end of the X-axis. If there is a stuck filament that needs to be manually removed this makes for difficult access if an enclosure is used.

### How to reproduce

Any filament change or pause command will initiate the new filament change park location.

### Expected behavior

I'd expect the printhead to be on the right end of the X-axis to allow for tools to retrieve a stuck filament. 

**Ideally, this park location should be a firmware option, so users can select the best location for their setups.**

chschmit, 2024-10-14:
Reflashing the firmware and rebooting resolved the issue.

chschmit, 2024-11-02:
The issue has re-surfaced for no reason. All the firmware and software are unchanged and now the printer is always parking the head to the left for a filament change which makes it difficult to access the side of the extruder if the filament is stuck.

tlachmann, 2024-11-06:
I also realized that change in the alpha FWs. If i was in the same situation I tried to put the printer into Pause becasue then the head is parked at the right side.
[The issue is located here](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/a810919b65b14e0a3e16da8f85a8992eb9a24efe/include/marlin/Configuration_MK4.h#L1593). There it is defined that for toolchange, M600, the head should be at x_min  +10
In FW before alpha it was defiend at [x_max -10, as defined here](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/f1de0fce157875165afc102b016f3f3fc84008ef/include/marlin/Configuration_MK4.h#L1588).

But, is it a bug or a feature... ;-)

Anther issue where the FW moved to stupidity.
It is totaly wired to move the head to a position, where access to the head is blocked.
But there are also other wired fw changes... like putting the Nozzle size settting another level deeper into the menu.
Or my favorite of stupidity:
If you load filament to nozzle via menu, the nozzle temp directly changes to 170°C. But if you unload the filament completely, the Nozzle temp keeps on the filament temperature e.g. 285°C for PC.

CZDanol, 2024-11-06:
Hey, I've inspected the code history for the M600 and it seems to be an oversight in changes. Thanks for reporting it.