prusa3d/Prusa-Firmware-Buddy #4253: [BUG] MK4S unexpected pauses during print

johann-miller, 2024-10-14:
### Printer model

MK4S

### Firmware version

6.1.3+7898

### Upgrades and modifications

None, but was originally a MK4.

### Printing from...

Prusa Connect

### Describe the bug

During two of my prints, I encountered pauses that were not inputted with the slicer (PrusaSlicer 2.8.1). 

In the first instance, the print pauses at the same place in the print that I did not add myself. It will park like a normal pause, and the option to resume shows up on the screen. The print continues as normal after selecting resume.

In another print, I had added a pause. It would pause normally, but immediately after resuming it would pause a second time, which would require me to hit resume again. This double pause happened several times, each with slightly different files (slight changes to the model). On one instance, the print "paused", but did not park, the nozzle remained in place in the print, requiring me to hit resume for it to keep going.

### How to reproduce

I'm attaching two binary gcode files that were giving me the issues. The issues are produced when printing the files (PETG)

Grip_0.4n_0.28mm_PETG_MK4S_19m.bgcode
 - This file has a pause from the slicer, but pauses twice in the same place.
 - Attached is a project file that generated this gcode, but has had a few changes since the code was produced
 
 mini13-all-frame project_0.4n_0.15mm_PETG_MK4S_3h21m.bgcode
 - This print randomly pauses on my printer in the same spot, even though there is no pause added in the slicer
 - I'm also attaching the exact project file in PrusaSlicer

### Expected behavior

The printer may have a random pause during print, pause twice in the same layer when only told to pause once, or the tool head will freeze in the print.

### Files

[Project files and gcode.zip](https://github.com/user-attachments/files/17357832/Project.files.and.gcode.zip)

CZDanol, 2024-10-14:
Hello, this should hopefully be fixed in 6.2.0-alpha2. Could you please try that one out and report if it still occurs?

CZDanol, 2024-10-14:
(whoops, sorry, accidentally pressed the close issue button)

DaveTexarkana, 2024-10-15:
10/14/2024 - I just had this happen twice in a row - am running firmware 6.2.0-alpha2 so don't think this is fixed.

This just started happening to me post Mk4S upgrade a week ago.  It pauses and then nozzle temp setting goes to Zero.  I was able to hit resume from the Connect Mobile app and it regained temp and started printing (albeit with a small blob). Last night a Benchy did the same thing (calibrating a new filament) and started again on it's own after 15 minutes - today 10/14/24 - anyone else seeing this? Thanks!

johann-miller, 2024-10-15:
> Hello, this should hopefully be fixed in 6.2.0-alpha2. Could you please try that one out and report if it still occurs?

Yes I will try the new firmware. Strangely enough, I've printed about 10 hours since posting this without any issues, so I will try to replicate it as best I can.

CZDanol, 2024-10-15:
Did it show any message before pausing?

DaveTexarkana, 2024-10-15:
No it doesn’t, shows the normal print messages while printing. I could
cancel/stop it, the LCD screen remains interactive, thanks!

CZDanol, 2024-10-16:
Hmm, what comes to my mind that might be causing this:
* Do you have touchscreen enabled? There might be some "ghost" touches on the pause button. Please try disabling.
* Maybe the knob might be faulty and creating false pressed. During the print, try rotating the knob so that the "stop" button is selected instead of the pause one.

DaveTexarkana, 2024-10-16:
O do have the touch screen enabled though I rarely use it, I will turn that
off. I did pu a ‘beauty ring’ from Printables on so I’ll double check
seating and try the rotation you suggest, thanks for the help!

loudaslife, 2024-10-19:
> In another print, I had added a pause. It would pause normally, but immediately after resuming it would pause a second time, which would require me to hit resume again. This double pause happened several times, each with slightly different files (slight changes to the model).

This particular bit sounds like the same issue as #4023. I had it happen several times, but never had any spontaneous pauses on prints that didn't already have pauses added. I can't reproduce it now, so I'm not sure whether it was fixed by 6.2.0.

It sounds like you're the only person who's had both the spontaneous pauses and the double pauses, so I would speculate these might be separate bugs and you just coincidentally experienced both.

johann-miller, 2024-10-23:
> Hmm, what comes to my mind that might be causing this:
>
>     * Do you have touchscreen enabled? There might be some "ghost" touches on the pause button. Please try disabling.
>
>     * Maybe the knob [...]

That's a good idea. I haven't encountered the issue in a while, but it could explain it. I believe @loudaslife could be right, and that I was experiencing two different bugs.

DaveTexarkana, 2024-10-23:
Thanks for the tip! I’ll try that!

CZDanol, 2024-12-03:
Internal ticket: BFW-6208

Prusa-Support, 2024-12-20:
This issue seems to be hardly reproducible so more specific testing conditions may hopefully help narrow the potential causes of the problem.

For instance, in case of future happenings please consider the following checklist.

- Is the problem reproducible when the printer is disconnected from Prusa Connect (disable connectivity)?
- Please check the wall socket - or plug extension/adapter if there is any - for proper grounding.
- Have you noticed static electricity phenomena in the environment around the printer?
  - Have you happened to sporadically experience shocks when touching the printer or other elements around?
  - Please double-check the PE cable for structural integrity and wiring firmness.
- If firmware upgrading (6.2.0-alpha2) didn't help, would the problem be reproducible after firmware downgrading instead?
- Are there third-party components connected/attached to the printer?
  - If so, please specify and try to reproduce the problem without such components.
  
Michele Moramarco
Prusa Research

Tomcus, 2025-02-19:
@patrickhammond Please try it with 6.2, there was an issue with automatic pausing that was fixed in 6.2, but we might not fixed all the issues.

Tomcus, 2025-02-19:
@DaveTexarkana Any luck with any of the suggestions?

DaveTexarkana, 2025-02-19:
It stopped happening, then I started getting “wrong filament loaded”
errors, now nominal behavior, very strange. Thanks for checking!

DaveTexarkana, 2025-02-19:
I have 6.2 loaded, previous response is current, thanks!

closed 2025-04-29 by github-actions[bot]