prusa3d/Prusa-Firmware-Buddy #4254: [BUG] First layer on MK4S is too low when using stock settings

BlueFyre, 2024-10-15:
### Printer model

MK4S

### Firmware version

6.1.3

### Upgrades and modifications

None

### Printing from...

Octoprint

### Describe the bug

The first layer when printing using stock profiles is too low on a large portion of the bed

### How to reproduce

Print a sheet on the bed (250x210x0.2) using 0.2mm SPEED and Prusament PLA settings

### Expected behavior

Perfect first layer

### Files

[image]

BlueFyre, 2024-10-15:
Also to note, I tried printing this with 215 C and 20 mm/s and the results were far better (but still not perfect)
[image]

frasier2k, 2024-10-20:
I can also confirm that after the upgrade to MK4S + HF Nozzle the first layer is too low. After setting a +0.04mm offset in the slicer it is ok. I used a satin sheet.

Shushuda, 2024-10-20:
I can confirm this. Here are my results, the broken corners are just testing layer strength, used Prusament PLA Azure Blue:

Smooth, 230C 40mm/s
[image]

Smooth, 215C 40mm/s
[image]

Smooth, 215C 30mm/s
[image]

Smooth, 215C 20mm/s
[image]

Shushuda, 2024-10-23:
I forgot to mention that my issues started after moving to Input Shaper firmware and profiles (which supports my findings regarding temp + speed settings). I've since then replaced the load cell, replaced the thermistor, tried multiple smooth sheets (I own 3), upgraded the printer to MK4S. Issue persists.

The only way to stop it in my case and get a perfect first layer is move back to 215C 20mm/s settings used before IS was a thing.

m00dawg, 2024-10-25:
Somewhat relates to https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3306 
That feature request would be a workaround by being able to set live Z on the printer and have it persist instead of having to put it in the slicer, which is what I'm doing now. That works but I'm worried I'll need specific values for multiple MK4s (which has prevented me from swapping any more MK3s to MK4s). Trying to dig through the source code for where I can simply add a load/save into the eeprom. Made some headway but not a lot.

Notably, it looks like Prusa tried to add in multiple sheets for the XL (and that feature *does* save live Z) but ran into issues based on code comments. I suspect that is one reason we don't have multiple sheets or savable live-Z offsets on the XL or MK4.

qlery49, 2024-10-27:
I got the same issue does someone know how to solve?
[image]

JoeScotty, 2024-10-27:
Same issue. Workaround by adding 0.05mm Z-offset in prusaslicer.
If there is some load cell result differences from individual loadcells, there should be a setting at the printer, where the offset could be stored permanently.

peterbenedicic, 2024-10-31:
Same problem, contacted Prusa chat, they sent me new load cell and hotend thermistor, completely rebuilt Nextruder with new parts, but issue with first layer rippling is exactly the same. As other already pointed out, its the same no matter which sheet or filament you use (including prusament). So probably not mechanical problem, something changed in firmware?

JoeScotty, 2024-11-01:
For getting some clarification I printed a a sheet with all original prusa profiles containing 3 square lines with a layer height of 0.2mm and extrusion width 0.8 (the only change against original settings).
Printer is an MK4S.

The print looks like 
[image]

Than I measured the thickness of the lines using a Mitutoyo micrometer getting a 7x7 result matrix.
Since layer heigth is set to 0.2mm, the values should be cloe to this value.
In the matrix i used µm instead of mm (0,2mm = 200 µm)

The result:
[image]

The matrix includes average per line and per column as well as the average over all: 0,167mm

I think, the offset is caused by load cell characteristics / bed pressure during test and just needs an offset.
The offset can either be set in Prusa slicer at all printer profiles (separate for every nozzle size)
But since its a hardware / firmware issue,  the correct location is the firmware.

I found the M206 command where offsets for X, Y, Z can be defined.
Tried to store this setting in the EEPROM, but at the next test print the EPROM value were set to default again.
Think the EEPROM content is kind of protected, since changing values without the proper knowledge can ruin your printer.

Spoke with prusa chat:
They know about this problem and a solution for entering the Z offset via display is in discussion.
And I should get an answer, if it is possible to change the offset.

andrem1987, 2024-11-06:
> Same problem, contacted Prusa chat, they sent me new load cell and hotend thermistor, completely rebuilt Nextruder with new parts, but issue with first layer rippling is exactly the same. As other already pointed out, its the same no matter which sheet or filament you use (including prusament). So probably not mechanical problem, something changed in firmware? And yes, +0.04 correction solves it. ![IMG_20241009_183723](https://private-user-images.githubusercontent.com/29711241/382071528-483593a7-21f0-4fe9-80cb-0088493aef07.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA5MzIxODMsIm5iZiI6MTczMDkzMTg4MywicGF0aCI6Ii8yOTcxMTI0MS8zODIwNzE1MjgtNDgzNTkzYTctMjFmMC00ZmU5LTgwY2ItMDA4ODQ5M2FlZjA3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTA2VDIyMjQ0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZhODM4ZmRjMzI2ZDdjODM2ZGUyNGM1NWRjYWMyZjZlZTEyMTc4NDZlMWZmYjJkYjI5NTkwYTJlMzEyZmRkOWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Sp_i46Q_4M42II2NqneQD8xZ4T3QEWevlBbjuMBDcYY)

Same for me. I contacted the support, they sent me a hotend thermistor and a load cell as well, but that didn't improve anything. Aftet setting a positive 0.045 Z-Offset the first layer was perfect again.
The possibility to set a persistent Z-Offset back again would be a good workaround for now.

JoeScotty, 2024-11-08:
@BlueFyre Interesting theory regarding the thermal expansion!

I just checked the magnitude of the thermal extension of brass using the calculator:
https://goodcalculators.com/thermal-expansion-calculator/
The entered values were a length of 20mm (hot zone), temperature difference of 50 °C and brass material.
The result was an extension of 0.0187mm, which is about half of the Z-offset value 0.04, which was posted above several times.
This explains about 50% of the cause of the to small first layer height!

I think, the workaround with entering a Z-offset in prusaslicer at printer settings (and save this a custom printer) is a better then calibrating with high nozzle temperature and having to deal with the ouzing and ripples.
The first layer MUST have the correct height as a first step!

The solution is the possibility to set a Z-offset in the firmware by Prusa.
Up to now I'm still missing the promised answer from the prusa chat technician.

Mike-FUT, 2024-11-11:
My first layers look the same. Doesn't matter if I use Prusament PETG, Pruasament PLA or any other non-Prusa filament. Flow calibration doesn't improve it.

Here's a picture of a PLA Silk print where it's extremely visible. With other materials it's the same but less visible.
[image]

Gman2606, 2024-11-12:
> > The only way to consistently get rid of the ripples for me is to reduce the first layer temperature by 10-15 degrees, depending on the filament. Raising the z-offset causes other problems in other parts of the build plate, like small holes next to the edge lines.
> 
> 
> 
> Reducing the first layer did seem to help a bit for me as well (as shown previously) but it's worth calling out that this creates added knock on effects. My other layer temperature is usually set higher than the profile (245 C) in order to avoid the glossy/matte filament transitions and the clog detection going off. Depending how fast it's printing, the temperature was too low to achieve the requested flow.
> 
> However, there's no waiting to transition from first layer temp to other layer temp which means it is more likely to go matte or clog. It's probably not waiting to avoid ooze in one part of the print
> 
> 
> 
> 
> 
> 
> 
> Re: Thermal expansion and offsets
> 
> If it is thermal expansion then it makes me wonder when that offset would need to be adjusted. For example, if we increase the print temperature, does it mean the offset changes?
> 
> Given the first layer speed we'd more likely run into over extrusion first for PLA. However, it also would be worth thinking about what happens when we swap the temperature for different materials
> 
> 
> 
> Anyhow just sharing my two cents and hoping someone smarter will come up with a better solution to all of this

About thermal expansion, offset will be calculated in comparison to the probing temp and added through out the whole print for example I'm printing at 230 the first layer so Z height=calculated offset. If I move to 240 in the middle of the print then Z height=new calculated offset+layer number sum.

I don't know if it's already happens probably does. But maybe someone did some bad math...

feyded1020, 2024-11-12:
I thought I was going crazy, I'm having similar issues with waves on my first layer. Using original parts and an ObXidian Prusa nozzle on my MK4S with Kingroon PLA on generic PLA preset. 

Printing temp is 230 for first layer, with 220 for all other layers. Scratching my head in regards on what can be done to achieve the claimed 'perfect first layer'.

mix579, 2024-11-12:
I really hope this is #1 on the list for Prusa software engineers as it complete destroys the myth of "perfect first layer". There's not one print anymore where I don't have to adjust live z manually. I don't know if it was upgrading from Mk4 to Mk4S or a recent firmware upgrade. But initially I didn't have problems with my Mk4 printers, at least with PLA (PETG always needed a
-0.012 adjustment). Now even PLA gives me the waves so nicely documented in this thread. The Mk3.5 I kept on the other hand--perfect first layers.

feyded1020, 2024-11-13:
Here's my current print at 225C with Kingroon PLA with ObXidian Non HF Prusa Nozzle on my MK4S. [image]
[image]
[image]

feyded1020, 2024-11-13:
Reprinted by changing offset +.04 as mentioned in this thread. This is result, and it really can print perfect first layer, just not out of the box like a Bambu can... Prusa please fix, or figure out where the discrepancy lies hardware wise!

[image]
[image]
[image][image]

The last pic was the only issue I saw, and that's where the first layer 'should' have started its print.

JoeScotty, 2024-11-13:
In my experience the most important parameter for reaching a good first layer is the thickness in the complete bed area.
The thickness will vary depending on the XY positions of the bed, since the bed is not really flat.
And bed levelling does not completely balance this.
A certain tolerance level is usually accepted.

Best reliable method I found is to print a square line test sample consisting of single lines using a smooth build plate.
If you have two or more lines side by side and the layer thickness is to low, the filament ist pressed upwards beside the nozzle tip and the measurement results in a too highvalue.
The line width should be around 0,8mm if using a 0.4mm nozzle to make sure the line is squeezed on top and bottom side
and have no higher shoulder.
Compare my post above.

Than measure the thickness using a good caliper or a micrometer and compare the value with the set first layer in the slicer.
If the thickness is in a sufficient tolerance field, then its recommended to play with other parameters like
extrusion rate, bed temperature, nozzle temperature, print speed, bed cleanliness.

mshipman, 2024-11-27:
I've been pulling my hair out over this exact issue for days, ever since performing the MK4 -> MK4S upgrade with the new HF nozzle. My MK4 always had perfect first layer performance, but after the upgrade I was seeing exactly the pictured rippling on every first layer (mostly using Hatchbox PLA with the provided profile in PrusaSlicer on a new, smooth PEI sheet).

Visually, I thought it looked like the classic first-layer-too-low scenario, but that's not supposed to be an issue with the load cell... so I spent hours dropping temperatures, tweaking the flow rate, slowing the printer down, and trying different materials (usually resulting in about 50% improvement) but the underlying problem still remained.

I finally relented and hardcoded the +0.04mm Z-offset in the slicer, and it's back to printing a perfect first layer, just like it did with the original non-HF nozzle.

It'd be great if Prusa could solve this all in software, but if that turns out not to be possible it'll be vital to have that option to hard code an offset in the firmware, again.

peterbenedicic, 2024-12-01:
@mshipman so now that you mentioned it, that with stock (normal brass nozle) works ok, got me thinking and indeed, it started with changing nozle to obxidian (before i did mk4s upgrade). So, I changed obxidian for stock one, preheat bed for 5 min, and now my first layer is near perfect again, repeatedly. So, I am wondering, what nozles majority of you use? Could it be, that obxidian is the culprit? Just for reference, this is now my first layer
[image]

mix579, 2024-12-01:
Interesting… You may one on to something. All my printers have Obxidian nozzles too.

On the other hand… I've been printing with Obsidian ever since they became available, never had a problem before the recent firmware updates.

Gman2606, 2024-12-02:
> @mshipman so now that you mentioned it, that with stock (normal brass nozle) works ok, got me thinking and indeed, it started with changing nozle to obxidian (before i did mk4s upgrade). So, I changed obxidian for stock one, preheat bed for 5 min, and now my first layer is near perfect again, repeatedly. So, I am wondering, what nozles majority of you use? Could it be, that obxidian is the culprit? Just for reference, this is now my first layer ![IMG_20241201_185926](https://private-user-images.githubusercontent.com/29711241/391353210-27f89dc8-fed1-479b-8bf0-f976b16b74dc.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxNjQxMjAsIm5iZiI6MTczMzE2MzgyMCwicGF0aCI6Ii8yOTcxMTI0MS8zOTEzNTMyMTAtMjdmODlkYzgtZmVkMS00NzliLThiZjAtZjk3NmIxNmI3NGRjLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjAyVDE4MjM0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRiNjRmYTgzYzIxZWY4ODE0ZGFhYTRhNzU2NmEzNzI3NjZiMTczMzY2OWVjMjdmNTFhYjY1ZTZlZjlkZGVhZDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.JC3albMrZUue6tVeE6YmlXczNcwKkPBMHKdqfbbejnM)

im using the the default brass HF nozzle issue is still there :(

Mike-FUT, 2024-12-02:
For me the nozzle type makes no difference. With the standard non-HF brass nozzle that I use on my MK4S results are sometimes terrible and someimes perfect.

I do have  guess though, even I can't confirm it:
When the bed is 100 % perfectly clean next print first layer looks perfect.
But during bed levelling (when not using MMU3) the nozzle always leaves these tiny amount of filament dots on the building plate that are really difficult to remove. And my feeling is that these small remainders of previous bed levelling cause the issues.
Currently I don't have my MMU3 connected and with MMU3 I don't recall ever having first layer issues.

But again... everything is only my "feeling". Will need to spend some time to confirm it.

mix579, 2024-12-02:
I didn't quite understand why I never had a first layer issue on my XL but now I checked and I noticed it actually doesn't have obxidian nozzles as I thought but good old brass.

simao, 2024-12-03:
I have this issue with all my nozzles, brass 0.4+0.6+0.4 HF and 0.4 obxidian

peterbenedicic, 2024-12-03:
Ok guys, tnx for input, it was just one of many thoughts regarding this issue. Would appreciate if anybody from Prusa staff could comment on this problem?

frnktrgr, 2024-12-07:
Same problem after MK4S upgrade. Z offset +.04 and again a perfect first layer. Nozzle not changed, as I also use the MMU3.

mix579, 2024-12-07:
The silence of the response to this issue is deafening. It seems that there are enough reports of this and similar situations here, in the Prusa Forums, and on Reddit to feel that this is not just a fluke that somehow affects a handful of people but something more profound.

lvpvsinfabvla, 2024-12-10:
MK4 same issue since upgrade MK4S and firmware. With PLA on regular, obxidian and HF nozzle. Will try lower the temperature by 10

jk1380, 2024-12-10:
> Same issue. Workaround by adding 0.05mm Z-offset in prusaslicer. 
Thank you so much, that fixed it for me too (as a workaround until Prusa fixes it). Had the same issue after my MK4S-upgrade and 0.20mm SPEED profile.

lvpvsinfabvla, 2024-12-13:
For me also set the live Z offset of 0.04 as a workaround make the first layer good again but introduces inconsistencies in the next layers that look alike underextrusion in some area/ direction and dimension inaccuracy in the model at the end of the print. This happen with different hotends, nozzles, plates, layer heights, pla and petg.

I m now following the suggestion of doing the bed leveling probe with the same or similar hotend temperature as the first layer.

Printers setting > start Gcode:
M104 ... 215)} ; set extruder temp for bed leveling
M109 ... 215)} ; wait for temp

manually remove oozle before probing

..so far the first layer is perfect again and I cant see inconsistency on the other layers that Z live offset would introduces

danopernis, 2024-12-13:
Hi folks, thank you for the reports and sorry for taking so long. I have created an internal issue BFW-6464 for investigation. Unfortunately I can't give you any estimate as to when we will be able to work on this, as we are currently finishing both CORE One and 6.2 release. In the meantime, please use the suggested Z offset workaround :pray:

BootsSiR, 2025-01-11:
MK4S @ v6.1.3

0.04mm offset on the left, default print on the right 

[image]

lvpvsinfabvla, 2025-01-14:
> Hi folks, thank you for the reports and sorry for taking so long. I have created an internal issue BFW-6464 for investigation. Unfortunately I can't give you any estimate as to when we will be able [...]

Be aware that for some of us, the Z live workaround introduces a series of issues in subsequent layers. However, matching the probing and nozzle temperatures resolves the problem flawlessly, suggesting correlation with between those temperatures needs further investigation

jonnieZG, 2025-01-18:
I am glad I'm not the only one. What puzzles me that I have contacted the user support and they actually sent me a new loadcell that I replaced. The problem with the new one is less pronounced, but still persists. Before I had to use 0.05 offset, which is now reduced to 0.03.

The issue definitely requires some serious investigation, and I am glad that @danopernis announced it being placed on the radar, but what puzzles me the most is why is that happening and is it really connected to the HF nozzle, or is it just a wear & tear of the loadcell that starts showing in time. Being a geek that I am, I am really interested in a technical explanation of this phenomenon.

mix579, 2025-01-18:
It's not just HF nozzles. Happens with my regular flow Obxidian nozzles as well, on all my Mk4S's. 

And I don't believe it's just a malfunction of the load cell. 

I'm flabbergasted that Prusa is not putting more resource on this. Given how they market the hell out of the "always perfect first layer" , the fact that a significant number of people seem to be having this problem that makes the Mk4 series less reliable than a Mk3 with a calibrated first layer, this should be numero uno on their list of priorities...

jonnieZG, 2025-01-18:
> It's not just HF nozzles. Happens with my regular flow Obxidian nozzles as well, on all my Mk4S's.
>
> And I don't believe it's just a malfunction of the load cell.
>
> I'm flabbergasted that Prusa is not [...]

I totally agree with that! Currently the only good thing is that the offset issue seems to be consistent, so it can be adjusted in the printer profile, but that's a pain, and not something that a new or an average user should be worrying about.

Even more, it should be more of a priority if the issue is present with all the printers, not just some. In either case, I sincerely hope this would not become the new "[issue #602](https://github.com/prusa3d/Prusa-Firmware/issues/602)" - and issue that was present with all the printers, but not all the users paid enough attention to it, so it has been pushed under the rug for quite a while.

jonnieZG, 2025-01-18:
I guess no one using a textured sheet would notice that issue, since a -0.03 to -0.05 offset is a recommended setting in that case. But for a smooth or a satin sheet it is a no-go.

TwinForce, 2025-01-19:
I really hope this issue gets resolved soon. I’m currently forced to print with a  0.03 mm Z-offset in my slicer profile.

I recently (21.12.2024) spent close to five hours (from 0:03am to 4:48am) in a chat session with support, and we went over the entire printer: we loosened and reattached screws, realigned brackets, remounted the heatbed, checked the belts and cabels from loadcell and so on. In the end, they sent me a replacement part—specifically a hotend heatsink with a loadcell sensor. After installing it, I unfortunately discovered that the problem still persists. It’s really frustrating. I was hoping that after upgrading from the MK3S+ to the MK4S, I wouldn’t have to spend so much time on bed levelling anymore…

For everyone who is interested:

Printer:		Prusa MK4S 
Add-ons:		MMU3 
Both are delivered as Kit.
 
Test scenario:
With MMU3 Installed and without. Both situations the printer needs the correction offset from 0,03mm for a perfect first layer.
Printed own sliced files and files from the support. We placed the object on different corners of the bed but no change of the problem.

Filaments tested with: 
	- Prusament PLA
	- Prusa PLA
	- Sunlu PLA+
	- Prusament PETG
	- E-Sun PLA

Firmwares that are checked out with the support with complete factory reset of the printer every time:
	- 6.1.3 updated from previous version
	- 6.1.3 complete factory reset
	- 6.2.0 Alpha

After getting replacing the loadcell with the spare part and the problem continued to occur, I removed and reassembled also the z-axis and y-axis to check if there was a problem.

@danopernis If u need information about my session with your support team for investigation, please let me know. :-)

Yog-Ourt, 2025-01-22:
Same issue here on a MK4 with default 0.4mm nozzle and 6.1.3 firmware.
I have better results with 0.05 offset change, but it's not perfect as before. It tends to be too high for satin sheet for instance.

I modified my start GCODE to put the nozzle temps to the same temp than my PLA profile, but i still have a bad result on center of my test object, but better results on corners.

When i compare the same print from MK3 and MK4...i'm sad. The upper layer is rough, compared to the MK3 printed part, where everyting is very smooth.

jonnieZG, 2025-01-23:
> When i compare the same print from MK3 and MK4...i'm sad. The upper layer is rough, compared to the MK3 printed part, where everyting is very smooth.

What you just said might imply problems with overextrusion! In that case, increasing the Z-offset would indeed help in mitigating the problems with the 1st layer, but the top layers, especially if 100% infill is used - would come out rough as you just described. 

Perhaps we are barking at the wrong tree here, as the problem actually might be overextrusion. Someone with a micrometer should investigate the actual thickness of the first layer (I just have a digital caliper which is not accurate enough).

jonnieZG, 2025-01-23:
> Whether it's over extrusion or something else, the fact remains that folks on this thread are seeing issues they didn't encounter with a Mk3S. It's also curious that some folks indicate success by adjusting the temperature while bed leveling, which speaks against over extrusion, at least not as the universal cause.

Most of the people don't properly insulate factors, or even perceive a problem, generally speaking. Same as it was with the issue 602 on MK3. Thickness, quality and dryness of the filament, temperature, other settings - any of that can affect the result. I noticed the issue after upgrading from MK4 to MK4S, using exactly the same filament. It could be due to the properties of the HF nozzle, but it also could have to do something with the HF profile in PrusaSlicer, or even a firmware version. I really hope that the good people in Prusa would investigate and solve the issue as soon as possible.

mix579, 2025-01-23:
Amen

Prusa-Support, 2025-01-25:
Thanks for all your feedback.

As highlighted in the thread, variables could be many, potentially more than the ones mentioned in this conversation.
We collected valuable pieces of insight from this conversation, that are compatible with our internal research, or that may open up other investigation paths. Thank you very much.

Minor improvements have been already worked on over time and as always, there is probably room for improvement.
Again, the variables are many and the investigation may take time. In the meantime, I would like to share a few guidelines that may be sufficient for satisfying results, or help cross out a few even trivial concerns.

> [!TIP]
>Large prints near the edges of the bed may sometimes be potentially tricky due to thermal gradients and dissipation - especially non-linear shapes, shapes with little contact area, rather sharp corners... 
>
> - For stronger adhesion (especially around the edges), you may slightly increase the bed temperature (∼ +5°C) to compensate for the stronger heat dissipation near the border (especially in cold environments). Excessive adjustment may lead to warping!
> - Another neat trick could be slightly lowering the first layer speed (~ -20%), especially when the first layer shape is "not so linear" (articulated, curved, segmented...).
> - However, in most cases, adding simple [brim](https://help.prusa3d.com/article/skirt-and-brim_133969) or [helper disks](https://help.prusa3d.com/article/shape-gallery_301453) on sharp corners is the best practice.
> - In edge cases, preheating the printer several minutes in advance allows for better heat absorption to help stabilize the geometry. This is generally negligible but I would recommend it for "challenging prints" (mentioned earlier), especially in cool or breezy rooms. Enclosures may help, once mildly warmed.

The investigation continues but I hope a few old tricks can improve your experience at the current time.

Michele Moramarco
Prusa Research

mix579, 2025-01-25:
Huh? We're not talking adhesion problems or warping here. So that "tip" is unfortunately utterly irrelevant. But good to hear that folks at Prusa are at least looking at it.

jonnieZG, 2025-01-25:
Thanks to Michele for the suggestions, but mix579 is right when he points out that this is not related to the issues addressed in this thread. 

Now let me introduce another confusing detail from the field, what just happened to me - the printer out of the blue - just started printing right, and I cannot reproduce the issue anymore!

What I did do:
1. The whole time I was printing, and when I first noticed the issue was while I was printing with PLA filaments. I was using several different filaments: green, white and black one - all of them were properly measured and their diameters and corresponding sweet-spot temperatures were entered in their own separate filament profiles. They were working great, until the point I just started noticing the issue and then I reported it. The issue was present with all of the filaments I tried (just tried those 3), and in each case entering a Z-offset value of 0.03 mm fixed the issue.
2. Yesterday I updated the printing profiles with the latest profile updates from Prusa. 
3. I tried printing with a new PETG filament I just purchased (255°C/85°C), and I used the latest default Prusa MK4S 0.4mm HF Nozzle printer profile - **and the issue was not there anymore!**
4. After that I tried printing again with my previous PLA filaments **and I couldn't reproduce the issue**. Tried several different printing profiles - all of them using the default Prusa MK4S HF profile with Z-Offset set to 0.0
5. Retrieved the old configuration from the backup - and still couldn't reproduce it.

What I did **not** do:

- Did not change the firmware
- Did not change the PrusaSlicer version
- Did not touch the printing plate
- Did not touch the nozzle nor any of the hardware
- Did not change the position of the printer or change any of its configuration

Go figure!

JoeScotty, 2025-01-25:
@ Prusa Support
Please have a look at my port from Nov 1st.
I printed a special test sample consisting of sindle squares with singele lines with a line width of 0.8, which are flattened by the 0,4mm nozzle tip.
Then I measured the thickness at 49 positions using a quality micrometer.
**The result shows there is a Z-offset problem.**

I tried to calculate the thermal extension of a brass nozzle between testing temperature and print temperature (The idea came from somebody else here). The result was that the exension seems to be part of the problem, but does not declare all the missing Z Offset.

I developed my own coreXY printer some years ago and learned to solve each problem at its root and not to hide it using other parameters which may have a slight influence on the impact of the problem.

Currently I help myself by manipulating the Z-Offset in the prusaresearch.ini.
Advantage: Just one edit affects all my printer profiles MK4S and MK4S MMU3.
Disadvantage: the change is gone after a prusaslicer upgrade and prusaslicer reacts very sensitively to errors in such changes.
After I made something wrong at my first steps even starting the installer again did not solve it.
Of course I can share but can not really recommend. So just a note for now.

qlery04, 2025-01-26:
As many have noted, a 0.04 Z offset helps achieve a perfect first layer. However, it introduces other problems in my case. While the first layer is perfect, I experience Y-axis layer shifting when printing at low layer heights like 0.1mm. These shifts disappear when I don't use the offset.

peterbenedicic, 2025-01-26:
> Thanks to Michele for the suggestions, but mix579 is right when he points out that his suggestions do not apply to the issues addressed in this thread.
> 
> Now let me introduce another confusing detail from the field, what just happened to me - the printer out of the blue - just started printing right, and I cannot reproduce the issue anymore!
> 
> What I did do:
> 
> 1. The whole time I was printing, and when I first noticed the issue was while I was printing with PLA filaments. I was using several different filaments: green, white and black one - all of them were properly measured and their diameters and corresponding sweet-spot temperatures were entered in their own separate filament profiles. They were working great, until the point I just started noticing the issue and then I reported it. The issue was present with all of the filaments I tried (just tried those 3), and in each case entering a Z-offset value of 0.03 mm fixed the issue.
> 2. Yesterday I updated the printing profiles with the latest profile updates from Prusa.
> 3. I tried printing with a new PETG filament I just purchased (255°C/85°C), and I used the latest default Prusa MK4S 0.4mm HF Nozzle printer profile - **and the issue was not there anymore!**
> 4. After that I tried printing again with my previous PLA filaments **and I couldn't reproduce the issue**. Tried several different printing profiles - all of them using the default Prusa MK4S HF profile with Z-Offset set to 0.0
> 5. Retrieved the old configuration from the backup - and still couldn't reproduce it.
> 
> What I did **not** do:
> 
> * Did not change the firmware
> * Did not change the PrusaSlicer version
> * Did not touch the printing plate
> * Did not touch the nozzle nor any of the hardware
> * Did not change the position of the printer or change any of its configuration
> 
> Go figure! 🤪

Hi, tnx for update. What about using old gcode, that exhibited issues? Is still present? Or it doesn't matter?

jonnieZG, 2025-01-26:
> Hi, tnx for update. What about using old gcode, that exhibited issues? Is still present? Or it doesn't matter?

I wanted to do that test first, but unfortunately, I deleted the old gcode! :(

Mike-FUT, 2025-01-27:
> Retrieved the old configuration from the backup - and still couldn't reproduce it.

Well, I sometimes get the advertised "perfect first layer" and sometimes a "terrible first layer".
And it can happen that i print the exact same item twice (g-code from printer) with the same material of course and one time it's perfect, the second time it's not.

A while ago I thought that a not perfectly clean bed might be the culptit since the nozzle adds these tiny dots to the levelling positions when a filament is loaded but I can likely rule that out now.

jonnieZG, 2025-01-31:
I just removed the trapezoidal nuts on my printer and compared to a spare one I had - the old ones have more play than the unused one. It could play a role in this issue.

Prusa-Support, 2025-02-01:
> Huh? We're not talking adhesion problems or warping here.

I hope you may not take this personal, but there may usually be very mixed issues reported in long threads.
I hope that those who are eventually willing to consider [my previous suggestion](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4254#issuecomment-2613934450) will benefit from these four simple tips in most common printing scenarios, at least before diving into advanced testing. You may still give them a try as a best practice: the last one is _not simply about adhesion and adhesion may even partially interfere with first-layer accuracy_ to some extent too.

Most importantly, please don't misunderstand the underlying message:
_thanks for the provided insights, this issue is currently being investigated, possibly in all its variants._

As an additional tip for community investigation, because temperature tests show some consistency, please consider trying to stick as much as possible to the same initial thermal conditions (printer and environment) when comparing results not so directly related to thermal factors like the print sheet, the material, the nozzle... Or even specific hardware parts like the trapezoidal nuts.

Michele Moramarco
Prusa Research

snrkl, 2025-02-02:
Interesting to note the Prusa support's notes (including yesterday's one). 

I too have been through the "replace all the parts of the nextruder" process.

What I can add to this is that I have a second Prusa MK4S that came out of the (factory assembled) box today, and has this exact issue.. right out of the box... a stock factory assembled Mk4S.

I really do hope that Prusa engineers are looking into this, as I have now spent 3 months troubleshooting, replacing parts, and not being able to print what I need to be printing: The first layer issues on large prints become filament blobs that end up causing collisions and layer shifts, and on the rare cases where it blobs off the prints, the print surface quality is terrible compared to my old MK3S+ (which is still going strong with the friend that bought it off me, incidentally... 😕)

qlery04, 2025-02-02:
@snrkl Mine also came pre assembled and had this issue already

qlery04, 2025-02-03:
@JoeScotty Could you tell me how to change it in prusaslicer.ini? I want to do that aswell.

akalurambo, 2025-02-04:
I did the return practises because I can't waste time with a new printer now it's been 30 days that I've been troubling to get my money back I lost 100 euros because they say that some items have been damaged absurd

JoeScotty, 2025-02-04:
> [@JoeScotty](https://github.com/JoeScotty) Could you tell me how to change it in prusaslicer.ini? I want to do that aswell.

Sure.
just want to add a little warning due to my experience, so it's on your own risk:

Have an MK4S MMU3.
Did some experiments with new MK4S-MMU3 printer types with additional nozzle sizes that are not officially supported.
I made mistakes in the prusaresearch.ini that caused Prusaslicer to no longer start. Uninstalling and reinstalling didn't help either. The cause was: Due to my mistakes in the prusaresearch.ini, Prusaslicer changed the prusaslicer.ini file, which then prevented it from starting.

The mod regarding Z-offset is less critical, but I still recommend backing up the following before making changes to the prusaresearch.ini:
- the complete Folder structure (which includes prusaslicer.ini):
  %APPDATA%\PrusaSlicer\
- the file
  c:\Program Files\Prusa3D\PrusaSlicer\resources\profiles\PrusaResearch.ini

Prusaresearch.ini contains sections for printer, print and filament. The blocks with information work with inherited properties. The inheritance chain sometimes goes over 5 levels.

Open   c:\Program Files\Prusa3D\PrusaSlicer\resources\profiles\PrusaResearch.ini using an text editor like notepad++
A block that inherits its properties to all printers in Prusaslicer has the heading
`[printer:*common*]`

one parameter in this block is
`z_offset = 0`
change it to
`z_offset = 0.04`

Spaces only before and after the =

If you have multiple printers configured and only one needs the modified z-offset setting, you have to look for another location for z_offset in prusaresearch.ini.

Not to forget:
In case of a prusaslicer update your mod is lost and needs to be redone afterwards.
And Prusa will for sure not support problems caused by modifiying prusaresearch.ini.

But they should finally solve this z-offset problem!

Hope this helps.

mix579, 2025-02-04:
Why not just set Z offset in Prusaslicer Printers>General>Size and coordinates instead of editing the .ini file? That change should be preserved across slicer updates. 

With the caveat that I just do the manual +0.04 adjustment when the print starts as I only see this issue with PLA filament.

JoeScotty, 2025-02-04:
Did this also as a first step.
The con argument, which counted for me is:
If you modify the printer profile and save it, you have one more profile in the printer list.
In my case (MK4S MMU3) I'm using MK4S profiles plus the MMU3 profiles. In total 12 printer profiles.
So I had to modify the Z-offset 12 times and- after saving - I have 24 profiles shown in prusasalicer.
Wanted to avoid this.

In my case I'm happy with the z_offset 0.04 for all filaments.

By the way: I think one part of the reason for this problem is the thermal extension of brass.
and the thermal difference between bed levelling and printing.
Some other guy posted this idea.

Lets calculate this:
The measurement nozzle temp is around 170°C
The printing temp is around 230°C
I'm not exactely sure where the expansion happens. Maxbe between Thumb screw and tip, which is a distance of about 30mm.
The formula for thermal expansion is
Le = L0 * a * Tdelta
with values:
L0=30mm,
a for brass = 18*10^-6 per Kelvin 
Tdelta = 60 K (230-170
Le = 30 * 18 * 10^-6 * 60 = 0,0324 mm.
That is at least in the region of the often stated 0,04mm z-offset. correction. I'm not sure about the L0=30mm.

If this is true, then a test scenario could be to perfomr bed levelling with 230 degrees (and clean the bed afterwards ;-)
And if this still supports this theory, one solution could be to precalculated the thermal expansion and the z-offset correction in the start gcode, because both temperatures and the nozzle thermal properties are known there.

[83 more replies]