prusa3d/Prusa-Firmware-Buddy #4255: [BUG] MK4s Failed Z homing when filament is loaded

FedericoLaTorre, 2024-10-16:
### Printer model

MK4s

### Firmware version

6.1.3

### Printing from...

USB Drive

### Describe the bug

After loading a fialament (in this case PETG) it seems the load cell detects some preasure that get detected as a colission on Z. Unload and reload the filament seems to solve the issue. Perhaps with a little retraction before the homing the preasure is relieve from the sensor.

### How to reproduce

Starting a print with a filament loaded.

### Expected behavior

To perform the home secuence successfully.

CZDanol, 2024-10-16:
Hello, please contact our support for this, this doesn't seem to be a FW bug.

FedericoLaTorre, 2024-10-16:
It is a function built in the firmware, I have no control over it, but it's a real problem and many people have it, only they posted on the forum. From my point of view a simple change to the macro loading the filament would solve it. After detecting som preasure in the load cell, retract some mm and thats it.

FedericoLaTorre, 2024-10-16:
Ok, so it´s very random, please leave this open and as soon as I experience the error I'll upload all the info, but the error is the detection of a collision on Z. The nozzle touch the bed (in the homming sequence at the start of a print), lift and then it doesn´t touch again, and keep trying mid air until the collision error appear.
I understand the preasure is in the oposite direction and I agree with you it should not detect a collision mid air, but it does something to do with the filament since if you unload and load it again the problem is fixed.

CZDanol, 2024-10-16:
What I can imagine is actually the reason would be the print head tugging the filament as it goes down and that would be causing the false Z triggers. That would indicate that there is rather a significant resistance for pulling the filament out of the spool and something should be done about that, because it could cause extruder skipping/underextrusion too.

Okay, reopening.

closed 2024-12-24 by github-actions[bot]