prusa3d/Prusa-Firmware-Buddy #4257: [BUG] MK4S firmware/gcode compatibility check: incorrect error

bryn51, 2024-10-16:
### Printer model

MK4S

### Firmware version

6.1.3

### Upgrades and modifications

MK4>MK4S upgrade

### Printing from...

Prusa Connect

### Describe the bug

When printer firmware>hardware>Nozzle is set to 0.25 size and bgcode sliced for mk4S HF0.4 Nozzle (obvious error) the error message on xLCD screen indicates a GCODE flavour error instead of nozzle size incompatible error.

### How to reproduce

a) set printer firmware > hardware>nozzle size to 0.25
b) slice a model to printer profile "Orignal Prusa MK4S HF0.4 nozzle, then send it to the printer via Prusa Connect.
c) observe what is the error message on the xbuddy screen

### Expected behavior

In the circumstances, a message indicating incorrect Nozzle size selected should be displayed.

### Files

[Front Door Moulding-2.3mf.zip](https://github.com/user-attachments/files/17388533/Front.Door.Moulding-2.3mf.zip)
[Front Door Moulding_0.4n_0.15mm_PETG_MK4S_12h4m.bgcode.zip](https://github.com/user-attachments/files/17388559/Front.Door.Moulding_0.4n_0.15mm_PETG_MK4S_12h4m.bgcode.zip)

CZDanol, 2024-10-16:
Hey, we've been doing some changes for the checks for the 6.2.0. Could you verify that the bug is still present in 6.2.0-alpha2?

H0Z0, 2024-10-16:
The error message is correct in 6.2.0-alpha2:

The G-code isn't fully compatible.
Nozzle diameter doesn't match.

CZDanol, 2024-10-16:
Aight, I'll close this as resolved then, if you're not objecting.

closed 2024-12-16 by CZDanol