prusa3d/Prusa-Firmware-Buddy #4259: [BUG] Loaded XL carriage crashes into parked tool

muesli, 2024-10-17:
### Printer model

XL

### Firmware version

6.1.3

### Upgrades and modifications

None

### Printing from...

PrusaLink

### Describe the bug

I cancelled a print after it had done its first layer, because I wasn't happy with the bed adhesion. The carriage (loaded with Tool 1) stopped at the back/right, just in front of Tool 5, when the print got cancelled.

I then selected "Park Tool" from the menu, which caused the carriage to ram tool 1 into tool 5 before recovering and parking the loaded tool 1.

### How to reproduce

I don't dare attempting to reproduce it, it looks like my tool 5 got damaged by this.

### Expected behavior

Never ram into another tool. The printer must have known the current X/Y position of the carriage and should prevent this.

muesli, 2024-10-17:
Sounds similar/related to #4154.

KamilKeski, 2024-11-10:
I experienced the same on the first print I tried with this version.  Tried rehoming to no avail.  It keeps crashing into tool 5.  I gave up on the second attempt as it's so violent I fear I've already caused damage.

CZDanol, 2024-11-11:
Hey, thanks for the reports. We have identified some motion-related issues in our 6.2.0 and they should hopefully be fixed in the next candidate firmware.

muesli, 2025-02-18:
Gotta hate that GitHub Action. Yes, that ticket is still valid otherwise I would have closed it myself.

CZDanol, 2025-02-18:
So this is still happening on 6.2.0 final?

KamilKeski, 2025-02-18:
I have not experienced this issue on 6.2.0.  I've been through many tool changes, rehomings, etc on the final release without this particular issue reoccurring.  From my personal experience this was easy to reproduce with pre-release and I cannot reproduce with release.

My $0.02 - I consider my issue resolved.

CptDelay, 2025-02-23:
For me it started to happen since the upgrade to 6.2.1. At least thats when i observed it for the first time. It seems to happen quite reproducible too. Whenever i cancel a Multi-Material Print when Toolhead 2 is active (2 Toolhead Version of the XL), it parks in front of Toolhead 1. Seems to be the same Spot where it parks after a succesful print, except it doesn't park the Tool first. When i command it to park the active Tool via the Control Menu it drives the Toolhead back into Toolhead 1 while starting its homing sequence. Only workaround is to manually release the Tool and dock it by hand.

dadinthegarage, 2025-02-25:
I am having the same issue with 6.2.0.

dadinthegarage, 2025-03-09:
I just had it happen again. Cancelled a print, 6.2.2, 5 TH. It was in the probing points mode, I cancelled print, it went back and crashed into tool 1. I did not get a crash error. Almost seemed like it went too far back. I then went to control and park current tool, went fine.

schrodlm, 2025-03-10:
I'll have a look

CZDanol, 2025-03-10:
FIY @schrodlm  there is a bug in aborting prints during the MBL that is fixed in the private. Could have been caused by that one.

alexandru-g, 2025-04-30:
I also have issues with the carriage crashing into parked tools. I've seen it in 6.2.4 for sure (just updated to 6.2.5 but I don't see it fixing the issue). The way it replicates (consistently) for me is this:

1. Abort print
2. Carriage moves to printer's back right (near tool 1)
3. Park current tool from menu. The printer will try and home, but something fails and it starts backing up into tool 1.

Randomly, after finishing a print, the next home is not performed "as planned". X or Y hits the gantry hard and the motor loudly skips and then it starts to move with those "unsure steps" on the X axis until it detects the axis start. Lets say this isn't such a big issue, but this morning it started doing thos "unsure steps" on the Y axis towards tool 1 again.

I don't believe there is something wrong hardware-wise with my machine, it works perfectly good in 99.9% of the time.

ssill2, 2025-12-04:
I can confirm I see this in 6.2.6 too.  I've not tried in 6.4.0 yet.  I might this weekend.  I saw other issues and and reverted back to 6.2.6 thinking it was a safer firmware.  I don't typically have to cancel prints often, but I was doing a tpu print with five colors and I tend to have to cancel prints to tune settings and reprint so this was a real bummer.

muesli, 2026-01-16:
Can also confirm this is still happening with 6.4.0.

closed 2026-09-14 by github-actions[bot]