prusa3d/Prusa-Firmware-Buddy #4260: [FEATURE REQUEST] Cycle through "Remaining time" etc. messages by touch

anghelos, 2024-10-17:
### Printer model

MK4

### Describe the feature

When I need to know how much time a print has left, I have to wait as the info on the LCD screen passes through all options ("Next change in N/A", "Printing time", ...).

It'd be great if we could go to the next message by tapping it.

### Expected functionality

Touching the 'Printing time' info area would change it to the next piece of information.

CZDanol, 2024-10-18:
Hey, I'd like to rework the print screen more significantly some day (BFW-5403), in a way that there would be none of this carousel stuff you have to wait for. Haven't had time to get to it yet, unfortunately :(

Tupson444, 2024-10-25:
>none of this carousel stuff you have to wait for

I agree, there is enough space on the screen that all of those values can be displayed at once.

stur0067, 2024-11-12:
Even just skipping the time until next change if it is NA would reduce the wait time. It seems to always be on that one when I want to see how much time is left.

neuron-whisperer, 2024-11-19:
I find this issue perplexing.

Every print cycles through four fields:

- Remaining Time
- Print Will End
- Next Change In
- Printing Time

100% of the time I look at my display during a print, I want to see Remaining Time. That's it - that's all I want.

But since this one field cycles through those four items over the course of 23 seconds (yes, I timed it), I'm usually stuck looking at the printer for 10-15 seconds on average before I get the one piece of information I want.

The truly baffling part is the mandatory inclusion of "Next Change In." 99% of my prints don't have any changes, so this field is almost always "n/a." Why show this useless field repeatedly on models that don't have any changes?

Think of all the ways that this could be solved:
- Display all four fields simultaneously since there is room to do so
- Provide some UI element to allow the user to scroll through it
- Allow the user to customize which fields are shown in the rotation
- Provide an option to show "Time Remaining" *somewhere* as a fixed field of the display other than this rotation

But, no, none of those are available. So I'm stuck standing at my printer looking at the LCD to find the one piece of info I need. It's an annoying speed bump in the otherwise smooth Prusa UI.

LarGriff, 2025-03-29:
👍🏻