prusa3d/Prusa-Firmware-Buddy #4261: [BUG] Setting pressure advance (M572) not working

welnanick, 2024-10-18:
### Printer model

MK4S

### Firmware version

6.2.0-alpha2

### Printing from...

USB

### Describe the bug

Setting pressure advance via M572 has no affect on print.

### How to reproduce

Run one of the attached gcode files with PLA filament, observe that the pressure advance value is modified throughout the print, but there is no effect on the resulting final print.

### Expected behavior

Changing the pressure advance value during the print should have a noticeable effect on the print.

### Files

[Example_Gcodes.zip](https://github.com/user-attachments/files/17427363/Example_Gcodes.zip)

welnanick, 2024-10-18:
I am going to downgrade to firmware 6.1.3 to see if the issue goes away, I will report back with the results.

welnanick, 2024-10-18:
Just confirmed that downgrading to firmware 6.1.3 resolved the issue, so the bug was introduced in either version 6.2.0-alpha1 or 6.2.0-alpha2 of the firmware.

CZDanol, 2024-10-18:
Hey, thanks for the report and for checking whether it worked in 6.1.3. I've created an internal ticket BFW-6216 and we will try to look in the issue.

ivous, 2025-01-21:
Hello,  
Fix is available and planned in the next release 6.2.

Problem is in the comments of the gcommands used:  
M572 S0.08 ; Set linear advance k factor

Parser was not working properly, the 'S' letter in the comments was replacing the argument value and therefore the proper value 0.08 was not taken into account.  
Comments were removed at the media prefetch level, however in the alpha release 6.2 this was changed and now comments are not removed for M commands, so the parser needs to handle them properly.  
Correction is done on the parser level.

Thanks again for the report and sample!

CZDanol, 2025-01-30:
Hello, 6.2 has been released: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0

The issue should be addressed, so closing. Have a nice day! :)

closed 2025-01-30 by CZDanol