prusa3d/Prusa-Firmware-Buddy #4266: [ENHANCEMENT] Play chime just before first layer calibration starts

murk-sy, 2024-10-24:
### Printer model

Mini+

### Describe the enhancement

When doing first layer calibration on a material with a high bed temperature (100°C being the obvious example), heating up takes quite a long time. It takes so long in fact that over time I've had quite a few instances where heating up (especially a custom thicker) bed sheet took so long that I only remembered to check it when it was too late to actually adjust the height accurately.

Additionally, because the first layer calibration is very slow, it can be very hard to hear the printer starting to move as well if it's not next to you but a nearby room instead.

### Expected functionality

1) Start first layer calibration
2) Bed heats up to target temperature
3) ABL runs and completes
4) Nozzle reaches target temperature
4) Play chime - long beep (same as "load filament now" screen) or 3 click beeps (a new "warning" chime?). Playing when the hotend actually starts moving makes the procedure consistent for all materials and the sound is played when something visibly starts happening
5) First layer calibration starts

I have considered:
- playing the chime 5-10° before the nozzle temperature is reached - would require some extra stuff to show on screen, otherwise the warning is played with no visual or movement feedback
- waiting for user input when nozzle reaches temperature - would require changing the procedure and manuals which is more work than it's probably worth, plus it might interfere with nozzle priming

This change is very minor and requires very little documentation changes, but it improves the QoL of first layer calibration a lot.

Tupson444, 2024-10-25:
I've added `M300` command right after  `M190 S[first_layer_bed_temperature]` in the start code for exact same reason (I have an old MK2 that takes a lot of time to heat to ~100°C)

CZDanol, 2024-10-31:
Hey, sounds like a neat and simple request. I've creataed a BFW-6267 internal ticket for it.