prusa3d/Prusa-Firmware-Buddy #4268: Hardware specs and limitations (100w heater)

Mulsk, 2024-10-25:
### Question

Hi I can see there is a software limitation in the firmware at 50watts for the nozzle heater.
Are there hardware limitations for this with the board? (It would be nice to know the hardware specs of the board).
(We have a Bondtech Lgx Mosquito hotend and a slice PT1000 sensor and a 100w heater and currently get the nozzle heater overcurrent warning because of the software limit. Can we adjust this to 100watts without burning out the board?).

If there are hardware limitations on the board is it possible to limit a 100w heater to 50watts in the firmware somehow or do we have to replace the heater to a 50watt? (do the bed and nozzle heater share the available current?)

CZDanol, 2024-10-25:
Hello, would you please specify what printer are you asking about?

Mulsk, 2024-10-25:
Yes it's an upgraded mk3s+ to 3.5 so essentially with mk4 electronics. But hotend is as specified before

CZDanol, 2024-10-25:
Okay, thanks. I have forwarded the question to the electro team.

Mulsk, 2024-10-26:
Tnx alot got the response that it is only designed to handle 50watts safely hence the limit in the firmware

closed 2024-10-26 by Mulsk