prusa3d/Prusa-Firmware-Buddy #4271: [ENHANCEMENT] G81 results in Prusa Connect

rbronco21, 2024-10-28:
### Describe the enhancement

Can G81 results be made accessible using Prusa Connect, rather than requiring a computer running Pronterface and a USB cable? I am using a Mini, but I'm sure the results would be appreciated for all models. A displayed heatmap could be nice, but a simple table or even the ability to send a command and see results would work.

https://forum.prusa3d.com/forum/original-prusa-i3-mk3s-mk3-improvements-archive/bed-levelling-mod-bed-level-calibration-guide-for-beginners-also/

CZDanol, 2024-10-29:
Hello, there is no G81 in the firmware.

3d-gussner, 2024-10-29:
Correct `G81` is Prusa MK2.5/S and MK3/S/+  specific gcode. `G29 T` does the same on Buddy firmware.

I can see in PrusaConnect for the MK3S under `Control` `Mesh Bed Leveling`. This is missing on the MK4 or XL and probably also for the MINI/+

This is something PrusaConnect team could look into.

@rbronco21 You can try to use `G29` followed by `G29 T` and use an online viewer by searching for "bed level mesh visualizer" in the interweb.

rbronco21, 2024-10-29:
I must have used G29 in the past. I was hoping to skip needing a PC or laptop next to the printer and grab the values through Prusa Connect. I don't see Mesh Bed Leveling for my Mini, but that does make me think the work is mostly done and just needs to be added to the Mini.

closed 2025-01-06 by github-actions[bot]