prusa3d/Prusa-Firmware-Buddy #4272: [BUG] Prusa XL - 6.2.0-alpha2+8702 - Toolheads heating up and not cooling down on single color print

fursphere, 2024-10-29:
### Printer model

Prusa XL

### Firmware version

6.2.0-alpha2+8702 

### Upgrades and modifications

None

### Printing from...

USB

### Describe the bug

Single color printer using tool head 1 - print job runs and completes fine.  Upon completion, tool head 1 cools down as expected.

However, tool heads 2,3,4,5 have heated up - even though they were not used - and remain hot long after print job has completed.

### How to reproduce

Run a single color print job using tool head 1 (first)

### Expected behavior

For a single color print - only the single tool head being used should heat up.  

After a print job completes, ALL TOOLHEADS should cool down under any circumstance.

CZDanol, 2024-10-30:
Hey, we've recently implement this thing, it should get into the next 6.2.0 release candidate :) Internal ticket BFW-5996

fursphere, 2024-10-30:
Safe guard fix might be to adjust slicer profile to tell printer to turn off all heaters, even if they weren't used during print job.   

Currently, it looks like slicer is trying to be intelligent using 'if' variables in end g-code section.   Just had set all tool heads to be temp zero after job?   remove the variables.

CZDanol, 2024-10-30:
That wouldn't work. The Gcode commands go through logical tool mapping, they do not control "physical" toolheads. As a result, you cannot control unmapped toolheads during the print from the gcode.

closed 2025-01-06 by github-actions[bot]