prusa3d/Prusa-Firmware-Buddy #4274: [BUG] Extensive heat creep into Nextruder causes filament grinding

ilka-schulz, 2024-10-30:
### Printer model

MK4 (pre-assembled) with original Prusa enclosure

### Firmware version

6.0.0+14794

### Upgrades and modifications

1. removed plastic cover from the LoveBoard, 2. using original Prusa 0.6 mm brass nozzle, 3. filament spool is held outside enclosure – issue is even worse with filament on original spool holder inside enclosure

### Printing from...

PrusaLink

### Describe the bug

This is a *hardware* bug but I could not find any repo for the MK4 hardware so I post it here.

### Observations
1. After printing one piece, the nextruder gets significantly warm. Using my thermal imaging camera, I was able to identify 51 °C roughly one minute after the print finished inside the print head close to the big gear wheel. The temperature in there seems to be more or less homogeneous so it is fair to assume that the big gear wheel pushing the filament is roughly 51 °C, too.
2. When printing several parts directly after each other, the printer will eventually struggle to push the filament through the nozzle.
   - The gear wheel grinds the filament and thus looses traction.
   - The nozzle does *not* clog.
   - For a couple layers, one can visually see how extrusion rates go down, then at some point the filament stops moving completely and the gear wheel grinds a chunk out of it.

### How to reproduce

1. print some ColorFabb HT @ 270 °C with any nozzle (0.4 mm / 0.6 mm original Prusa brass nozzle)
2. use the original Prusa enclosure
3. use Input Shaper "Structural" mode
4. print for at least 3 h without cooling down in between

### Conditions
These are my specific conditions. You might be able to reproduce the issue with slightly different settings, too. Especially if your ambient temperature is higher than 21 °C.

| condition | value |
| --- | --- |
| temperature inside enclosure | ≈ 31 °C |
| temperature outside enclosure | 21 °C (controlled by AC) |
| humidity outside enclosure | usually 60 % RH |
| Filament | [ColorFabb HT](https://colorfabb.com/ht-dark-gray) (some cristalline PET) |
| Filament feed | spool hangs outside enclosure to reduce filament temperature |
| nozzle diameter | 0.6 mm |
| nozzle temperature | 270 °C |
| bed temperature | 105 °C |
| print time | 121 min per print + heating, etc., I am printing many pieces in a row|

### Expected behavior

- printer prints part without grinding filament
- print head / nextruder gear wheel does not get that warm, especially not 51 °C

neofir3, 2024-10-31:
Hello, I had this problem with my MK4 in the enclosure a lot when I was printing ABS and the doors of the enclosure were closed, as was the lid.
Then I always got the message "Filament blocked" because I had activated the option. But I didn't always notice it straight away. By the time I noticed it, the extruder had probably cooled down a bit.
Then the filament was unloaded and reloaded and then it worked again.
I think it got a little better with the S upgrade.
At least 1 ABS print of almost 7 hours went through in the enclosure without any problems.

tlachmann, 2024-11-06:
I also had that issue, indeed, it is temperatur related but will be triggerd by the idler screw tension. I was investigation here last two months by myself.
Found some design issues with Mainplate, that needed to high idler tension and finally grinded filament.
But also found some other improvements that helps to get rid of filament grinding (stuck filament), that can be done by yourself.

If interested, you can find the solution and explaination to the issues and root causes [here](https://www.printables.com/model/1059347-mk4mk4s-with-mmu-main-plate-with-improovement-step).

I `m printing since nearly 2weeks, >2500 filament changes and > 60 hrs of printing time without any stucked filament issue. I also printed 2 models with more than 24hrs each, in 70°C heated chamber.

closed 2025-01-13 by github-actions[bot]