prusa3d/Prusa-Firmware-Buddy #4275: [ENHANCEMENT] Be able to set minimum and maximum print fan range to override dynamic cooling

stur0067, 2024-10-30:
### Printer model

MK4

### Describe the enhancement

If my gcode uses layer-time dependent cooling or dynamic cooling, the gcode constant overrides my manual print fan speed changes. If there was the ability to set a lower-bound and upper-bound (defaulting to 0% and 100%) I could more easily make on-the fly adjustments.
For example, If I have to cancel some objects, my layer times might be too short and I'd like to just force a higher fan speed. I can do this by setting the minimum to 60%. Any command in the gcode that tries to issue a fan speed below that value is ignored. This means dynamic cooling or cooling for bridging that request higher fan speeds would still work.

### Expected functionality

The forced minimum and maximum print fan values (default of 0% and 100%) would allow better control despite having a g-code that issues many fan speed changes. They could be a separate entry from the normal print fan setting and maybe be called "Print fan speed range forcing".

stur0067, 2024-10-30:
Maybe this should actually be a feature request

Tupson444, 2024-10-31:
In Marlin firmware, there is a feature called Extra Fan Speed that is similar to this idea, but more limited. It allows setting two different fan speeds, e.g. normal printing and bridging, which can both be adjusted during print without resetting every time a new fan command is executed (the gcode file needs to be adapted in order to work).

Your proposal would work even when using dynamic fan speed, and wouldn't need gcode file to be specially prepared for it. But min/max fan speed still wouldn't allow increasing max fan speed.

How about having a way to generally increase or decrease fan speed (like speed %)? E.g. set "general %" to 110%, and all fan commands will be increased by 10% (capped at max fan speed of course). Although it would be a bit confusing to have two percentages (actual % of the fan and "general" %).

stur0067, 2024-10-31:
Yes, being able to set a delta to apply would be useful and it would
preserve dynamic fan speeds having some control. I was under the assumption
that the gcode does not provide any information to know why the fan speed
is changing (like for bridging). If there is no way to distinguish then you
just have to move the minimum fan speed override high enough to get you
what you want.

murk-sy, 2024-11-07:
Tangentially related issue, could be implemented in the same/similar way as Tupson444 said:
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3343

Handling it as an offset would likely make it easier to understand and make menus more consistent to use

Tupson444, 2024-11-07:
>Handling it as an offset would likely make it easier to understand and make menus more consistent to use

I agree, an offset to fan % would be the best way to correct the fan settings on the fly

Tupson444, 2025-01-14:
not stale

stur0067, 2025-01-18:
Thanks for keeping this alive....
I continue to run into cases where this feature would be nice to have. Even if it is just the ability to force a guaranteed minimum fan value that can not be decreased below or just a global +10% modifier, for example. Its much more common to run into this type of situation rather than one where you need to prevent the fan from going above a certain level.

Tupson444, 2025-03-24:
still not stale ...

Tupson444, 2025-05-30:
again, just commenting to prevent closing

stur0067, 2025-05-30:
Thanks for keeping this alive. At this point I'd be happy with just a fan speed override that causes all additional fan speed changes in the gcode to be ignored