prusa3d/Prusa-Firmware-Buddy #4277: [FEATURE REQUEST] enable X Y homing in tuning menu.

kiklhorn, 2024-11-01:
### Printer model

MK4S

### Describe the feature

I accidentally shifted the Y position when solving a problem with the filament.

### Expected functionality

If rehoming is possible then I can continue printing after accidentally shift. Now I must cancel the print.

tlachmann, 2024-11-06:
Alternatively you can send GCode "G28 X Y" if you using Octoprint, I dont´t know if on prusa connect, this is possible.

I needed that in the same behavior of stucked filament in extruder but after long time of investigation I got rid of filament stuck issue. I postet everything [here](https://www.printables.com/model/1059347-mk4mk4s-with-mmu-main-plate-with-improovement-step). I needed to redesign the Mainplate, becasue the orignal for MMU has a design error, but also some other points are need to imrpoove to get it running smooth.

Since nearly 2 weeks and more then 2500 filament changes and more as 70 hours of printing, no filament stuck issue.

You also need to know, that after 120 sec of Printer Pause, X and Y Axis will be shut down, due to firmware setting.

I have learned here, that the only possible way is to find a solution by yourself becasue as with the most reported issues, it will be closed and goes to nirvana.

CZDanol, 2025-01-06:
Interesting idea. Created the BFW-6535 ticket.

nicolai-wachenschwan, 2025-04-10:
I would also like this Option, as it would have saved some prints with layershifts. (Of cause there are ways to solve the Problem, but they are not easy and take Time and precision.