prusa3d/Prusa-Firmware-Buddy #4278: [BUG] Missing Warning Step For High Flow Nozzles and Cold Pull

Tripp-Sautner, 2024-11-01:
### Printer model

MK4s

### Firmware version

6.1.3 & 6.2 Preview

### Upgrades and modifications

High Flow 0.4 Nozzle

### Printing from...

Cold Pull

### Describe the bug

Cold Pulls are not effective with high flow nozzles and can lead to PLA getting stuck behind the filament guides in the gear cluster.
[image]
That said Filament
[image]
[image]

"Successful" Cold Pull w/ High Flow Nozzle
[image]

### How to reproduce

Use High Flow Nozzle
Perfom Automatic Cold Pull

### Expected behavior

Use High Flow Nozzle
Perfom Automatic Cold Pull
Get Warning noting high flow nozzle not supported
Cancel/ Continue messagebox

### Files

[image]

Tripp-Sautner, 2025-01-05:
This is defintly still important, I dont think there is a lot of clarification about High Flow not being allowed to automatically cold pull. This is enabled in software to be allowed but can cause damage.

CZDanol, 2025-01-06:
Hello, our conclusion (BFW-6138) is that the cold pull is generally working fine for the CHT nozzles, and thus we do support it and will not be adding any warnings.

closed 2025-01-06 by CZDanol