prusa3d/Prusa-Firmware-Buddy #4279: [ENHANCEMENT] Turn off part cooling fan during M600 filament swaps.

Sembazuru, 2024-11-02:
### Printer model

MK4S

### Describe the enhancement

Right now with the new MK4S fan strength and direction when I'm doing an M600 filament change the ramming extrusion and the purge extrusion is blown towards the back. My MK4S is in the Prusa enclosure so I sometimes loose the extruded filament towards the back of the printer and sometimes into what has already been printed, especially with the reduced access of reaching in through the enclosure doors. If the fan stopped blowing (maybe turn off when the extruder starts moving towards the filament change location so it has time to spin down) for the filament change I can more easily manage filament change related extrusions.

### Expected functionality

Ideally, the fan would stop when the extruder starts heading towards the filament changing area so it has time to spin-down. Then once the filament change is done, start the fan to the previous speed while the extruder is heading back to the print so it has time to spin back up to the speed the slicer set for that portion of the print.

Sembazuru, 2024-11-04:
FYI, this request stands as of v6.2.0-alpha2. This has been a long standing issue since MK3 OG, but the MK4S blower makes it worse (because the direction and strength compared to both the MK3 line and the MK4 OG side blower), prompting me to finally submit an issue on it.

leprikon77, 2025-01-07:
No, you're not 👍

CZDanol, 2025-01-07:
Hello, this is actually already requested in #1223 (BFW-5474). Bumping the priority.

Sembazuru, 2025-01-07:
My google-fu must have failed me. I did try searching for this but didn't find #1233. Thanx for the confirmation that it is (slowly) being addressed in the dev tracker.

Sembazuru, 2025-09-15:
No problem. Closed with fix implemented in 6.4.0-alpha.

closed 2025-09-15 by Sembazuru