prusa3d/Prusa-Firmware-Buddy #4280: [QUESTION] Input shaping Calibration option is not visible in menu

Mulsk, 2024-11-06:
### Question

Hi we've bought the Accelerometer Set to Calibrate input shaping. (on our Mk3.5s prusa printer.)
But the Calibration option is not shown in the menu under Input Shaping on our custom built firmware.

Are there some special build flags or modes needed to enable this option?
we're using: [--preset mk3.5 --generate-bbf --build-type release --bootloader yes --final]

(The Calibration option is visible in the official firmware but we cannot use that to completion since we have a different heat sensor than stock).

[image]

Mulsk, 2024-11-06:
I think I found something in ProjectOptions.cmake

# MK3.5 disabled until we figure out how to handle cable switching
set(PRINTERS_WITH_INPUT_SHAPER_CALIBRATION "MK4" "XL" "XL_DEV_KIT")
  
if(${PRINTER} IN_LIST PRINTERS_WITH_INPUT_SHAPER_CALIBRATION AND BOARD_IS_MASTER_BOARD)
set(HAS_INPUT_SHAPER_CALIBRATION YES)

Mulsk, 2024-11-06:
We solved it by adding MK3.5 to the list

closed 2024-11-06 by Mulsk

CZDanol, 2024-11-07:
Seems like you weren't basing your builds on the 6.2.0 branch/label - that option should be enabled for MK3.5 there.

Mulsk, 2024-11-07:
> Seems like you weren't basing your builds on the 6.2.0 branch/label - that option should be enabled for MK3.5 there.

Ahh.. tnx for the tip we just took it from the master branch not the labels.