prusa3d/Prusa-Firmware-Buddy #4282: [ENHANCEMENT] Filament sensor status in API / metrics

samdurnxn, 2024-11-06:
### Printer model

All

### Describe the enhancement

There are many usefull informations in the API and in metrics to monitor in different ways the printer but there is 2 missings things.
I search in the API, serial information and in the available metrics but I can't find the filament sensor status and for the MMU3 the Finda status, it will be great to have it.

If these informations are already available, sorry . Just tell me where I can find it. 

### Expected functionality

Add the information in the printer status information in the json return will be useful or if it's much simple to implement, add it in the metrics like this it will be activate on demand by add a M331.
Thanks in advance

CZDanol, 2024-11-07:
There are fsensor_raw, fsensor, side_fsensor_raw metrics already. Not sure about the finda, maybe "side_fsensor" will work?

samdurnxn, 2024-11-07:
I activate all metrics regarding filemant sensors but I think it's not those I need.

In my gcode : 
M331 fsensor_raw
M331 fsensor
M331 side_fsensor_raw
M331 side_fsensor

I received in syslog : 
"fsensor_raw,n=0 v=1873381i 57157"
"fsensor_raw,n=0 v=1871784i 15851"
"fsensor_raw,n=0 v=1871755i 80585"
"fsensor_raw,n=0 v=1871624i 38462"

Only this metrics when the extrusion block is moving on my MK4S.

Maybe the information of filament detection is only send to the screen for information and not send elsewhere. What do you think ?

samdurnxn, 2024-11-10:
I search in the code and I found that the values are available in the marlin_vars() object with the functions mmu2_finda() and active_extruder() but not use in the API or the metrics. 
I don't know if it easy to add it in the metrics ? if yes, is it possible to add it in the next version of the firmware ?

samdurnxn, 2025-01-14:
Hello, it will be great if you can add these informations in the API or metrics. Can you confirm if you can add it in your todo list ?

closed 2025-03-24 by github-actions[bot]