prusa3d/Prusa-Firmware-Buddy #4283: [BUG] Screen flickering during print

edy80y, 2024-11-07:
### Printer model

MK4S

### Firmware version

6.2.0 alpha2

### Upgrades and modifications

None

### Printing from...

USB Drive

### Describe the bug

~80% into a 4hr print the screen started to flicker. Printing continues as per usual, only the screen is affected. Screen issues range from flickering, to black screen, to horizontal flip, to negative colours. (see pics)

Note: I flashed 6.2.0 alpha2 hoping the flickering from 6.1.4 would go away.  It was OK for a short while, but flickering is present in this firmware too.

### How to reproduce

1. Buy a MK4S
2. Print something

### Expected behavior

1. The screen should never flicker.
2. See Step 1

### Files

[image]
[image]
[image]
[image]
[image]

murk-sy, 2024-11-07:
From my experience with the Mini screen, this is more indicative of a poor/loose connection than a firmware issue. I recommend unplugging and replugging the cables going from the screen to the mainboard.

edy80y, 2024-11-07:
Thanks for your input guys.

I had a chat with Prusa Support team about this yesterday and they are sending me a new screen and cable.  

At first I thought this was software related considering there was no flickering until after boot loader and displaying FW on screen. 

When the screen arrives (in a few days) and I can test that it works OK I will have this bug closed.

neofir3, 2024-11-10:
I also have the problem that my xLCD flickers, it looks very similar to the pictures above.
I only got my new xLCD from Prusa a month ago.
Because the first one was a bit too far to the right on the board, and you couldn't see everything on the right edge properly.
The new xLCD worked great for the first few weeks.
After updating to 6.1.4 it started flickering. :-(
I only did the update to 6.1.4 because it was displayed directly in PrusaConnect.
[image]
[image]
[image]

edy80y, 2024-11-15:
New screen arrived, swapped it over and no more flickering !! I will keep an eye on this over the next few days, then have this big closed.

neofir3, 2024-12-07:
So I bought a new xLCD cable from Prusa at my own expense.
My MK4S is in a Prusa Enclosure. I put the xLCD cable from the BuddyBoard directly to the xLCD without any major fastening or anything else and only as a test.
It's been working like this for about 2 weeks now WITHOUT a single problem with the xLCD.
But the strange thing is that the flickering first occurred when I restarted the MK4S during the upgrade from 6.1.3 to 6.1.4.
I didn't have it before. But let's see how long it works now. I still think it's partly due to 6.1.4 but I can't prove it for sure.

Mike-FUT, 2025-01-30:
> Hello guys, we have released a FW version 6.2.0-RC1 with the option to reduce display refresh speed: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-RC1

Hi @CZDanol,

You linked the RC firmware.
I guess it should have been the release?
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0

Also, feedback from my side: I received a new display weeks ago which I replaced and since then no single glitch, not with 6.1.3, not with 6.2.0 RC1.

And one more hint: You might need to add an option to activate/disable the new display refresh speed option to Prusa Connect because - at least when my old display had the issue - it was completely unusable. No chance that I could have selected any option at all. Mine went stable after being on for an hour or so but I'm not sure if all displays do. If they don't it will be tricky to select anything.

Also the problems with my display started from one day to the next. No warning (like slight initial issues or so), nothing.

CZDanol, 2025-01-31:
Yes, sorry, should have been the release.

THanks for the tips.

bca3244, 2025-05-08:
Any updates on this issue? I have a new Core One with this issue... changing the setting worked for now. I contacted support and they are reluctant to replace any parts, they want me to check all of the cabling (which I have).

ottobonn, 2025-07-24:
I also have a new Core One with this issue. The display worked fine for the first few weeks, then started flickering, inverting, going blank, etc. I am not aware of anything that changed between when it worked and when it didn't.

I just updated to firmware 6.3.4, which mentions "Automatic display refresh rate fallback" in the release notes. When the printer started up, it said "Display issues detected" and prompted me to reduce the refresh rate. The screen is working with the reduced refresh rate now.

Is this an electrical problem with the display cable? Does it need shielding?

p-Wave, 2025-07-25:
> [@p-Wave](https://github.com/p-Wave) Please try to set the Settings -> Hardware -> Display Refresh Speed to LOW

Done. Will get back to you on whether it works.