prusa3d/Prusa-Firmware-Buddy #4284: [QUESTION] Status of E3D Hotend support (specifically mk3.5)

mghenderson, 2024-11-07:
### Question

I currently am working with a mk3.5 (which has been upgraded twice I believe, mk3s -> mk3s+ -> mk3.5S) and I've been having issues with E3d's hotends.  I know they are not yet fully supported so I have an original hotend coming from Prusa soon I hope, but I am still hoping to understand where everything stands.  

I've tried both the E3D Revo quick-change (Prusa model) and the E3d v6 (also prusa model) and both times I'm getting an overcurrent error as soon as I attempt to set a temp on then nozzel.  As I still have the original PSU, which I believe is 12V as compared to the new 24V available on the store, both of those hot ends were the 12v models.  I've tried changing the hotend type in the firmware, I've tried stock, I've tried everything I can think of.  I would have just given up given the whole "not-supported" issue, but I've seen multiple comments in issues on the repo saying they are using them just fine.  What am I missing?

LoakAudio, 2024-11-09:
Hi,
Referring to [this page](https://help.prusa3d.com/article/multimeter-usage_2117), MK2/2S is 12v and MK3/3S/3S+ is 24v, hot end and heat bed.

mghenderson, 2024-11-16:
Well, that would pretty much cover it.  I don't know how I missed that, & I very much appreciate the help.

closed 2024-11-16 by mghenderson