prusa3d/Prusa-Firmware-Buddy #4285: [BUG] MK4 with firmware 6.1.* crashes on some files that worked with 6.0.* task:IDLE? error

David-Fabijan, 2024-11-07:
### Printer model

MK4

### Firmware version

6.1.3, 6.1.2, 6.1.0

### Upgrades and modifications

None

### Printing from...

USB drive

### Describe the bug

After upgrading (quite late) from the 6.0.4 version of the frimware to the 6.1.3 version I've noticed that for some of the .bgcode files that worked previously the system now crashes with a blue screed BSOD and an task:IDLE? error

### How to reproduce

Update the frimware to the 6.1.3 version and try to print the attached gcode file. 
[NinaPot_0.4n_0.2mm_PLA_MK4IS_6h49m.bgcode.zip](https://github.com/user-attachments/files/17658157/NinaPot_0.4n_0.2mm_PLA_MK4IS_6h49m.bgcode.zip)

### Expected behavior

You should end up with a 3D printed flower pot. Instead a preview appears (as expected), but instead of starting to print, the printer restarts into a BSOD with the task:IDLE? error.

CZDanol, 2024-11-07:
Hello, for us to debug this, please:
* Post a screenshot of the BSOD
* Send us a crash dump

You can also just try 6.2.0-alpha2, it's likely that this has already been fixed.

David-Fabijan, 2024-11-07:
Here is the image. I will try the 6.2 version right away.

[image]

David-Fabijan, 2024-11-07:
It seems that version 6.2.alpha works fine. The error is gone. There does seem to be a slightly longer loading time for the file to be read than with version 6.0.4, but the printer software no longer crashes and it prints normally. Thank you for the suggestion.

CZDanol, 2024-11-07:
Thanks for the BSOD screenshot. From the description, I do believe that this is indeed a bug that we've fixed in 6.2.0, so I'm closing this ticket as done :)

closed 2024-11-07 by CZDanol