prusa3d/Prusa-Firmware-Buddy #4287: [BUG] XL unintentionally retracts filament after filament change when in move axis

Raymogon, 2024-11-09:
### Printer model

Prusa XL 2T

### Firmware version

6.1.3+7898

### Upgrades and modifications

Enclosure + Highflow Nozzles

### Printing from...

Prusa Connect and USB-Drive

### Describe the bug

When you do a filament change and then immediately go to manual Axis movement on the printer display the printer automatically retracts the filament the same amount that it was pulled in from the filament change, this for me has lead to multiple occasions where the filament blob at the end caused the extruder to clog up. It doesnt happen every time for me it is a 50/50 Chance it happens, dont know why. It even happens if the extruder has cooled down after filament change and you go to move axis and to extruder and tell it to heat up, after it has heaten up it also immedietly retracts the filament.

### How to reproduce

1. Go to: Filament, Change Filament, select tool
2. Go to: Control, move axis
3. if Nozzle isnt heated up anymore tell it via the extruder move command

### Expected behavior

The extruder should show 0mm if you havent moved it yourself  und should not retract anything if not commanded.

RPospisil, 2024-11-15:
Hi, I spotted the same issue. 
It seems that it is the issue for a long time - I spotted it on 6.1.3 (and older) + also on 6.2.0-alpha2.

JordyDH, 2024-11-26:
We see the same behavior on our XL machine.

p27182, 2024-12-24:
I just saw this issue in slightly different context-  if i stop a print and go into the movement menu after as im starting to troubleshoot what went wrong, the extruder starts to move in reverse for some time.   this is on 6.1.3

p27182, 2024-12-24:
just tried reproducing this on 6.2 alpha2 and it does not seem to be present...

CZDanol, 2025-01-02:
@p27182 thanks for the test. If someone reports this on 6.2.0, we will look into it, otherwise we will ask you guys to either update to 6.2-a2 or wait for the next release candidate.

RPospisil, 2025-01-02:
Hi,

it is happening on 6.2.0 alpha 2 (as I mentioned in my previous post, btw it happened several times to me in last few days).
BTW I 'found workaround':
- before you go to the Control->Axe movement (ovladani->pohyb osy) do scroll between all icons on the main menu - just use knob full scroll full left and then full right. Note touch screen cannot be used here as it is not possible to 'scroll' between icons.
- I tried the workaround several times and it is 100% reliable for me. 
- My typical flow is to 'load filament' and then do extruder retraction -5mm + cooling to avoid filament splitting. So If I go directly to axe movement right after filament loading, then it happens in most cases. If I use the workaround, then it won't happen.
- when I observed this issue for the first time (fw 6.1 and may be older), it was happening for extruders 2 and others, not for 1st extruder. Now it is happening for all extruders.

CZDanol, 2025-01-04:
All right, created an internal ticket BFW-6529.

RPospisil, 2025-02-03:
> Is this issue still present for you in firmware version 6.2.0? I tried to reproduce it but couldn't. Let me know if you're still experiencing the problem.

Hi, it seems that it issue is fixed - tested on 6.2.0.

schrodlm, 2025-02-11:
OK, closing this issue for now, feel free to open it again if the issue reappears.

closed 2025-02-11 by schrodlm