prusa3d/Prusa-Firmware-Buddy #4290: [FEATURE REQUEST] Eject MMU "Empty" filament

Glaedr304, 2024-11-10:
### Printer model

MK4 MMU3

### Describe the feature

Imagine you have just finished a MMU print, you were running low on your first spool of filament and you setup your second spool as a joined spool. Your first spool ran out as you intended, and the PrusaMK4 switched to the second spool with ease. You want to now go load a new filament in the first bay, and to your disgruntlement, you have no way to retrieve the filament remnant.

### Expected functionality

I think a good solution to this would be for the MMU to have an eject function, where the multiplexer goes all the way to the right (henceforth referred to as the 6th slot), and the MMU then ejects whichever bay (1-5) that the user input. 

The filament would be released out the front of the MMU for the user to retrieve. And because the mmu multiplexer is in the 6th slot the filament is not routed to the hotend. This operation can be done when the hotend is cool.

CZDanol, 2024-11-10:
That almost sounds like the Eject Filament feature that we have implemented in the Filament menu

Glaedr304, 2024-11-10:
that's truly incredible that you were able to add this feature so quickly, and upload the software to my printer without my doing anything. Amazing.

closed 2024-11-10 by Glaedr304

CZDanol, 2024-11-11:
It was actually a lot of effort from my side. I had to sneak into your house during the night and update the printer. Also, I've cooked some lasagna for you and left it in the fridge.