prusa3d/Prusa-Firmware-Buddy #4292: [BUG] MK4S/6.1.3 "Length of an axis is too long"

Mike-FUT, 2024-11-11:
### Printer model

MK4S

### Firmware version

6.1.3

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

Sometimes, when the printer wrongly detects a crash at the beginning of a print it tries to check the axis before resuming the print but this always fails with the message "Length of an axis is too long". This is on an un-modified MK4S.

### How to reproduce

- print a relatively high item
- this will cause the head to be rather high on the Z-axis
- now, when using a filament with high friction in the teflon tubes and starting the next print the load cell will recognize the pull of the filament when moving to home position and it will go down in many steps, every time a bit lower until the head reaches the building plate. if these steps are getting too many I guess the display will show a "crash detected message". (btw. this going down in many steps due to the filament back pull with higher friction is not the issue in this ticket. this is not good but I guess that's just by design when homing with the load cell sensor)
- when clicking "retry" on the display after this message the print head will fullly move x and y axis
- this will never work and it always gives the "Length of an axis is too long" message

Behaviour can be seen here:
[https://photos.app.goo.gl/AroAY1xNppxqciKW9](https://photos.app.goo.gl/AroAY1xNppxqciKW9)

panneff, 2024-11-13:
Have you contacted Prusa support regarding this?

Propheta13, 2024-11-14:
Have same issue. Calibration passes, but start of the print always recalibrates X axis and may detect "crash" if started to high on Z-axis.

Mike-FUT, 2024-11-14:
> Have you contacted Prusa support regarding this?

I have, here with this bug report.
Since normal calibration via menu works perfectly I guess it's just wrong values in the firmware for Y-axis length during this rare recalibration procedure.
Just a guess though but I think this is a simple bug with numbers in the firmware.

Mike-FUT, 2025-01-15:
I'll update this one when 6.2 is released to see if that has been fixed but so far I haven't seen anything in the releasenotes ofr 6.2 alpha/RC that would indicate that issue being fixed.
@CZDanol, not sure if the video that I created is sufficient for the firmware bug to be found. Let me know if you need more...

closed 2025-03-24 by github-actions[bot]