prusa3d/Prusa-Firmware-Buddy #4293: [BUG] M21 does not respond properly

timur-tabi, 2024-11-11:
### Printer model

MK3.5s

### Firmware version

6.1.3+7898

### Upgrades and modifications

None

### Printing from...

USB

### Describe the bug

When an SD card is inserted, the M21 command is supposed to respond with one of the following:

-     SD card ok
-     Card successfully initialized

However, all it does is:

```
2024-11-10 09:29:37,617 - Send: M21
2024-11-10 09:29:37,637 - Recv: ok

```
This makes Octoprint think that the SD card is not available, even though it is.

timur-tabi, 2024-11-11:
See also https://github.com/OctoPrint/OctoPrint/issues/5077 for the debug trail.

timur-tabi, 2025-01-13:
It's still important to me.

timur-tabi, 2025-01-13:
I'm assuming that if the printer support USB but not SD, then the M21 command should work with the USB card as if it were an SD card.  So on the MK3.5, M21 should respond with "Card successfully initialized".

bkerler, 2025-01-25:
I wrote a PR #4428 for this issue.

CZDanol, 2025-01-27:
PR merged. Should appear in the 6.4 version.

closed 2025-11-18 by Prusa-Support