prusa3d/Prusa-Firmware-Buddy #4295: [BUG] MINI reports sheet Z height uncalibrated when minimum value is set

jenda122, 2024-11-13:
### Printer model

Mini

### Firmware version

6.1.2+7894

### Upgrades and modifications

None

### Printing from...

USB drive, Prusa Link

### Describe the bug

When Z offset calibration is set to a limit value, e.g. -2.000mm, the printer thinks Z offset has not yet been calibrated. "Please complete Calibrations & Tests before using the printer." is shown when attempting to start print.

I believe this is caused by the following expression in `src/common/SteelSheets.cpp:SteelSheets::IsSheetCalibrated`:

`return std::isfinite(sheet.z_offset) && sheet.z_offset < zOffsetMax && sheet.z_offset > zOffsetMin;`

where `zOffsetMin` seems to be set to `Z_OFFSET_MIN`, which is defined to be `-2.0F`. Therefore, if you dial the GUI to the limit value (-2.000), the condition will return false.

`sheet.z_offset <= zOffsetMax && sheet.z_offset >= zOffsetMin` should probably be used.

### How to reproduce

Set Z height for the current sheet to -2.000 (the lowest value can dial). Try starting a print.

### Expected behavior

The printer will accept the calibration value.

CZDanol, 2024-11-14:
Hey, thanks for the report. Looks like you're pretty much correct, good job :) Created an internal ticket BFW-6334 for this.

CZDanol, 2025-01-09:
Hello, this should now be fixed in https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-RC1.

Closing the issue. If the problem persists, please let me know and I will reopen it.

closed 2025-01-09 by CZDanol