prusa3d/Prusa-Firmware-Buddy #4296: [BUG] Est. Finish Time shows "Heat Bed Tile Nr"

tmssr, 2024-11-14:
### Printer model

MK4

### Firmware version

6.1.3+7898

### Upgrades and modifications

None

### Printing from...

USB-drive / Prusa Connect

### Describe the bug

Occasionally, the estimated Print to Finish Time displays "Heat Bed Tile Nr" instead of a time. It does happen random, for example today while printing the stock benchy from usb drive.

Sometimes the clock is seen for a fee seconds, but then it switches back to the told message.

Note: German Version and German Message
 

### Expected behavior

It should display a time of when the print is finished

### Files

[image]

CZDanol, 2024-11-15:
Hey, thanks for the report. Will be fixed. Internal ticket: BFW-6343

JimBim89, 2024-12-18:
Another user had the German text of msgid "Heating disabled due to 30 minutes of inactivity." shown there.

CZDanol, 2024-12-18:
The fix should get out in the next firmware :)