prusa3d/Prusa-Firmware-Buddy #4298: [FEATURE REQUEST] Add ability to switch extruder when filament spool is empty

3DBob99, 2024-11-15:
### Printer model

MK4S + MMU3

### Describe the feature

When spool empties, the MMU nicely has short piece of filament to be removed from the front. Then to continue printing a new spool needs to be loaded in the same extruder position. This can be inconvenient when spools are in a dry box. I know of spool join, but that has to be set up prior to start of print.

This is similar to another request, [FEATURE REQUEST] continue printing with another tool head #4054, but that looked like XL specific. Please close this as a duplicate if appropriate.

### Expected functionality

When spool is emptied, instead of requiring that same extruder slot to be used, ask user if they want to refill same slot, or switch to a different extruder.

CZDanol, 2025-01-15:
Internal ticket: BFW-6621

LarGriff, 2025-03-29:
I still think it’s a great idea.

Prusa-Support, 2025-06-05:
Thanks for this input.
This will actually be handled as a duplicate of #4054. They are marked with the same internal ticket.
This issue may also partly find common solutions with issue #4432 (tool-mapping during the print).

Michele Moramarco
Prusa Research

closed 2025-06-05 by Prusa-Support

3DBob99, 2025-06-05:
Excellent! Thank you.