prusa3d/Prusa-Firmware-Buddy #4300: [FEATURE REQUEST] Low-power idle mode triggered by gcode

timur-tabi, 2024-11-18:
### Printer model

MK3.5s

### Describe the feature

I would like there to be a low-power idle mode which is triggered by a gcode command.  It would have the following features:

1. LCD is off
2. All heaters and fans off (so firmware will wait until nozzle is cold)
3. Serial port is disconnected (so Octoprint doesn't try sending command)
4. Processor is still on, but firmware is running in an idle loop.  So memory has not been cleared, but the processor is not really executing code.
5. PrusaLink is shut down.

The only way to exit low-power mode is to either power-cycle the printer, or press the button on the control panel.

The purpose of this is to allow the printer to wake up quickly for the next print, but still consume almost no power if idle for hours.  I would add the gcode command to the End G-code, so that it triggers automatically at the end of the print.  If I ever need to print anything else, I can just press the panel button when I remove the print.

### Expected functionality

.

CZDanol, 2025-01-21:
Hello, this is an intereseting idea. I've created an internal ticket BFW-6640 for this and will pitch the feature to the product - absolutely no promises though :)

timur-tabi, 2025-01-21:
That's fantastic, thank you!