prusa3d/Prusa-Firmware-Buddy #4301: [BUG] Enclosure fan not running after print completion

zehkaiser, 2024-11-18:
### Printer model

XL 5T

### Firmware version

6.1.3+7898

### Upgrades and modifications

None

### Printing from...

USB via Prusa Link

### Describe the bug

The enclosure fan turns off immediately following print completion despite the Post Print Filtration option being enabled with Post Print Filtration Duration set to "5 min." The heat break fan shows "100% / stuck," but I'm not sure if that's normal without the enclosure. You can hear the fan spin down in the video.
[image]

https://github.com/user-attachments/assets/70138df4-3a5d-4dcf-81fd-d96a76ded7db

### How to reproduce

Enable the enclosure filtration fan and enable Post Print Filtration with a duration of 5 minutes. Allow a print to complete.

### Expected behavior

Enclosure filtration fan should run for 5 minutes before shutting off.

CZDanol, 2024-11-18:
Hey,
would you mind trying to update the FW to 6.2.0 to check if it hasn't been already fixed?

Also, the heatbreak fan should definitely not be saying it's stuck, I'd recommend checking it out. Does it do that for other tools, too?

zehkaiser, 2024-11-19:
I updated the firmware and printed on Tool 2. Both tool fans now say "0% / stopped," but the enclosure fan still shuts off as soon as the print has completed. Tomorrow I'll run another print on Tool 1 to see what the fans say after the print completes.

Ma2002rio, 2024-12-21:
Hi, I can confirm. I have the same problem with version 6.1.3. The filter fan switches off immediately after printing, despite the set run-on time.

NHodgesVFX, 2024-12-28:
Same here on v6.1.3.

zehkaiser, 2025-01-23:
On 6.2.0-RC1 I found that this is still a problem. I don't know how I did it, but I was printing with a tool that was not 1 and the fan stayed on for the 5 minute duration. It only happened once, and I couldn't reproduce it.

CZDanol, 2025-02-03:
Hello guys, we will be likely rewriting the filtration system (to also accomodate for the CORE One), so this should be fixed during the rewrite.

CZDanol, 2025-04-07:
Hello, I have been continously working on porting features from the XL enclosure code to the new unified chamber filtration. They should all have been ported (testing pending), moving the XL to the new codebase is yet to be done. Not sure if I'll make it in time for 6.4.

Ma2002rio, 2025-04-07:
> Hello, I have been continously working on porting features from the XL enclosure code to the new unified chamber filtration. They should all have been ported (testing pending), moving the XL to the new codebase is yet to be done. Not sure if I'll make it in time for 6.4.

Thank you for the update! 👍🏻 Will be eagerly waiting for it 😉

CZDanol, 2025-04-07:
Thank you for your patiance :)