prusa3d/Prusa-Firmware-Buddy #4302: [ENHANCEMENT] Show printer name in Prusa Link

neuron-whisperer, 2024-11-20:
### Printer model

All

### Describe the enhancement

This one is really, really simple.

I own two printers (a MK4S recently upgraded from a MK3S, and a MK4 with an upgrade kit incoming). When I access Prusa Link via Wi-Fi to feed them models, the page shows me a lot of stuff about the printer... *except its name.* I have to check out the IP address to make sure I'm talking to the right printer.

Can you please just put the name of the printer somewhere on the page? Thanks.

danopernis, 2024-11-20:
By name I assume you mean printer model? There is also a hostname, which the browser will happily show in the address bar :wink: Other than that I don't know about any other _name_ you are referring to.

This can be implemented but our schedule is really tight right now. We are willing to accept patches though. Note that this probably needs to be implemented also in the https://github.com/prusa3d/Prusa-Link-Web repository.

neuron-whisperer, 2024-11-20:
Oh! Great - I'll work on it.

Printers have user-assigned names stored in firmware. Mine are named Protoceratops and Protohadros, and when PrusaSlicer offers to send a model directly to a printer, it shows me a status page *including names,* like this:

[image]

I'm merely asking for the same information to be displayed in the PrusaLink page. Should be very easy as the printer itself is serving the page!

I understand that you can't get to it; I'll be happy to roll up my sleeves and see if I can implement it.

bkerler, 2024-12-14:
This is already integrated into Prusa-Link-Web, it just hasn't made it into the firmware yet.

neuron-whisperer, 2025-04-02:
I'm still interested in this and bumping it to keep it in the queue.

bkerler, 2025-04-13:
@neuron-whisperer PrusaLink is already showing the printer name (hostname) :

[image]

Or do you mean something different ?

neuron-whisperer, 2025-04-13:
My dashboard has no hostname field.

[image]

bkerler, 2025-04-13:
Ok, interesting. For me, I see the information (using the latest 6.2.4 rc1). Are you using the latest fw ?

neuron-whisperer, 2025-04-13:
Both printers are on 6.2.3+8862. I don't run release candidates but full releases. Maybe it's being added in 6.2.4.

bkerler, 2025-04-14:
Yeah, could be. Let's wait for the release, then we can figure out if the name is displayed.

closed 2025-07-09 by github-actions[bot]

3d-gussner, 2025-08-15:
FYI: It should be fixed in next major release.

paulcobbaut, 2025-11-03:
Same issue here. I now have two MK4 printers on DHCP and no way to see the hostname in Prusalink. Firmware is up to date.