prusa3d/Prusa-Firmware-Buddy #4303: [BUG] MINI+ stops responding to mDNS queries after a while

dzuelke, 2024-11-20:
### Printer model

MINI+

### Firmware version

6.1.2

### Upgrades and modifications

None

### Printing from...

PrusaLink via Ethernet

### Describe the bug

After some time into a print, my MINI+ will no longer be reachable under its mDNS hostname (in my case `prusamini.local`) - it simply does not respond to the mDNS multicast query anymore. It still responds fine to pings using its IP address, and the web interface remains accessible using the IP address.

A quick fix is to disable and re-enable the Ethernet connection (e.g. by switching to WiFi and back, or by getting a new DHCP lease).

This doesn't appear to be happening when the printer is just sitting idle - it only occurs after a while during a print.

Even after the print has concluded, there are still no responses to mDNS queries once the printer is idle again; the only fix is, again, to re-init the network connection on the printer.

In my testing, if I simply run a `dig -p 5353 prusamini.local @224.0.0.251` in a `sleep 60` loop or similar while the MINI+ is printing, then it will correctly keep working forever, so I cannot say after how long of a print time this happens.

It seems that the background process that handles mDNS responses "goes to sleep" after some time during a print and then never wakes up again, as this method of "keeping it awake" using periodic mDNS queries prevents the problem from occurring.

### How to reproduce

Print a longer print (say, two hours), then attempt to resolve the printer's IP address using its `${hostname}.local` address.

### Expected behavior

Multicast mDNS query responses by the printer do not stop after a while when printing.

dzuelke, 2025-01-08:
Another observation: with a freshly booted printer, I used PrusaSlicer to send a job via PrusaLink (the printer is set up using its IP address).

During the job upload, I tried accessing the web interface using the mDNS name, and that didn't work (hangs forever to load, since the name does not resolve).

Using the Tune menu to reset the network settings somehow (e.g. interface on/off or "edit" the hostname), the web interface then came up immediately.

CZDanol, 2025-01-08:
Hey, care to test it on 6.2.0-alpha2? Might have gotten fixed already.

dzuelke, 2025-01-08:
(and https://github.com/prusa3d/Prusa-Firmware-Buddy/commits/master/lib/WUI/mdns has had no changes in a while)

CZDanol, 2025-01-08:
I'm not sure, but this issue might not be necessarily related only to the mdns library.

dzuelke, 2025-03-10:
Yup, this is still happening and still dearly important to at least me ;)

CZDanol, 2025-03-11:
Have you checked the 6.2 or not?

dzuelke, 2025-03-11:
Yes, sorry, I should have clarified.

CZDanol, 2025-03-11:
Okay, thank you for clarifying. I've created an internal ticket BFW-6899 for this.

dzuelke, 2025-03-11:
I'm currently (far) away from the printer, but once I'm back, I can try and make a "proper" reproduce case for it. The tricky bit is figuring out after how long of a time it starts happening, and it needs to be during a print, but maybe a gcode file that just pauses for a certain amount of time is enough to trigger it.

CZDanol, 2025-03-11:
That might be very helpful, thank you :)

dzuelke, 2025-04-04:
Okay so I don't even have to start a print anymore on the latest firmware. My freshly booted MINI+ didn't respond to mDNS queries. I re-set the network connection (toggle Ethernet to Wi-Fi and back, or go into hostname edit and just confirm, or get a new DHCP lease, or whatever), and it worked. I let it sit for a few minutes, and it was broken again until a re-set. Then it was stable for at least a longer period of time than the first time around.

Here is the log; `dns-sd` thankfully prints timestamps:

```ShellSession
$ # freshly booted mini, no mDNS response:
$ dns-sd -G v4 prusamini.local 
DATE: ---Sat 05 Apr 2025---
 0:04:41.989  ...STARTING...
^C

$ # just to prove that mDNS works in general on the network:
$ dns-sd -G v4 brother.local  
DATE: ---Sat 05 Apr 2025---
 0:05:21.173  ...STARTING...
Timestamp     A/R  Flags         IF  Hostname              Address           TTL
 0:05:21.295  Add  2             14  brother.local.        192.168.1.21      240
^C

$ # (re-set the network on the mini)
$ dns-sd -G v4 prusamini.local
DATE: ---Sat 05 Apr 2025---
 0:06:30.115  ...STARTING...
Timestamp     A/R  Flags         IF  Hostname              Address           TTL
 0:06:30.116  Add  40000002      14  prusamini.local.      192.168.1.61      120
^C

$ # still works:
$ dns-sd -G v4 prusamini.local            
DATE: ---Sat 05 Apr 2025---
 0:07:40.758  ...STARTING...
Timestamp     A/R  Flags         IF  Hostname              Address           TTL
 0:07:40.760  Add  40000002      14  prusamini.local.      192.168.1.61      120
^C

$ # aaaaand it's broken again after a few minutes:
$ dns-sd -G v4 prusamini.local
DATE: ---Sat 05 Apr 2025---
 0:10:51.572  ...STARTING...
^C

$ # re-set the network again...
$ dns-sd -G v4 prusamini.local
DATE: ---Sat 05 Apr 2025---
 0:11:30.020  ...STARTING...
Timestamp     A/R  Flags         IF  Hostname              Address           TTL
 0:11:30.021  Add  40000002      14  prusamini.local.      192.168.1.61      120
^C

$ # still works:
$ dns-sd -G v4 prusamini.local
DATE: ---Sat 05 Apr 2025---
 0:13:09.925  ...STARTING...
Timestamp     A/R  Flags         IF  Hostname              Address           TTL
 0:13:09.927  Add  40000002      14  prusamini.local.      192.168.1.61      120
^C

$ # still works:
$ dns-sd -G v4 prusamini.local
DATE: ---Sat 05 Apr 2025---
 0:16:54.800  ...STARTING...
Timestamp     A/R  Flags         IF  Hostname              Address           TTL
 0:16:54.898  Add  2             14  prusamini.local.      192.168.1.61      120
^C
```

dzuelke, 2025-04-05:
Hours later, it's still happily responding:

```ShellSession
$ dns-sd -G v4 prusamini.local
DATE: ---Sat 05 Apr 2025---
 2:16:59.385  ...STARTING...
Timestamp     A/R  Flags         IF  Hostname                               Address                                      TTL
 2:16:59.499  Add  2             14  prusamini.local.                       192.168.1.61                                 120
^C
```

dzuelke, 2025-04-05:
And still happily up 16+ hours later:

```ShellSession
$ dns-sd -G v4 prusamini.local
DATE: ---Sat 05 Apr 2025---
18:54:39.621  ...STARTING...
Timestamp     A/R  Flags         IF  Hostname                               Address                                      TTL
18:54:39.702  Add  2             14  prusamini.local.                       192.168.1.61                                 120
^C
```

dzuelke, 2025-04-05:
And now, just as I was about to upload a test gcode file that simply sleeps, it no longer responds:

```ShellSession
$ dns-sd -G v4 prusamini.local
DATE: ---Sat 05 Apr 2025---
19:07:10.033  ...STARTING...
^C
```

dzuelke, 2025-04-07:
Tested with a simple "dwell" GCode:

```gcode
M107
G4 S3600
```

And it failed once, but not a second time:

```ShellSession
$ # started `G4 S3600` job on printer
$ sleep 60m; dns-sd -G v4 prusamini.local
DATE: ---Sun 06 Apr 2025---
22:20:25.769  ...STARTING...
^C

$ # re-set network and started `G4 S3600` job on printer
$ sleep 60m; dns-sd -G v4 prusamini.local
DATE: ---Mon 07 Apr 2025---
14:50:47.424  ...STARTING...
Timestamp     A/R  Flags         IF  Hostname                               Address                                      TTL
14:50:47.507  Add  2             14  prusamini.local.                       192.168.1.61                                 120
```

I know that e.g. mDNS runs as an "app"; can these "apps" crash? Do they get evicted or suspended under high system load during a "real" print? Is it maybe possible that e.g. the network interface or the mDNS app goes into some sort of sleep state that would necessitate a call to `mdns_resp_restart` (see
https://github.com/lwip-tcpip/lwip/blob/ca0395c5ae54273b4fd542ef25b57d4b759678f8/doc/mdns.txt#L61-L63)?

Are there any logs I can pull from the printer somehow?