prusa3d/Prusa-Firmware-Buddy #4304: [BUG] Repeatable Tool Crash During Complete Individual Objects and Multimaterial Painting

TheOneTrueRazor, 2024-11-21:
### Printer model

XL Multitool

### Firmware version

6.2.0-alpha2+8702

### Upgrades and modifications

None

### Printing from...

USB Drive/Prusa Connect

### Describe the bug

The toolhead is crashing in to the 4 instance on my print sheet as seen in the included photos. This seems to be happening only when I am using the Mulitimaterial Paint function. I have tried printing this arrangement before and had an identical crash. Today I simplified my models to basically create a single color version, a shell version for the lack of a better description, and the print worked for instance 5. I included a photo.

### How to reproduce

Send the multicolor GCODE to the printer.

### Expected behavior

What I believe is happening is once instance number 4 is finished, tool #5 is parked, and the next tool, #1 is picked up. (This is the tool for layer 1 on instance 5.)  The XL is not dropping the Z to allow for the tool to clear instance number 4. 

That required Z movement is missing from the GCODE. Instances 1-4 printed fine because the tool head wasn't traveling over existing objects when tool changes were made. Instance 5 is too close to the dock and the only path is over an existing object. That is why I am having this failure. I believe this is confirmed by my successful single color/tool print.

[Hornets Archive.zip](https://github.com/user-attachments/files/17851087/Hornets.Archive.zip)

TheOneTrueRazor, 2024-11-22:
Can confirm this is what is happening. The bed is rising to first layer height before the nozzle is where it needs to be to start printing. The order of print commands should be:
1. Nozzle moves in to position for the first layer.
2. Bed moves to the correct Z height.

I have video of this, but it is too big to upload here. It's pretty self explanatory, but please let me know if you would like me to either compress it or upload it to YouTube. Thanks.

CZDanol, 2024-11-23:
Hello, thanks for the report. I'm not sure yet if this should be resolved on the printer side or on the slicer side, I will discuss it with the slicer team :)

TheOneTrueRazor, 2024-11-25:
Thank you. I feel better knowing I wasn’t the only one who was confused. I wasn’t sure which to post this in. 😁

If it needs to be moved please let me know if I need to do anything to move it.

CZDanol, 2024-11-25:
Hey, talked a bit through it with the slicer and it seems that it will either be a purely firmware fix, or some sort of collaboration. I will have a proper meeting tomorrow.

Razor412, 2024-12-03:
FYI, I changed my user name. Hopefully I didn't mess anything up by doing that. Let me know if I did. ;-)

3d-gussner, 2025-11-19:
Please try firmware [6.4.0-RC2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.4.0-RC2) and report back.

Razor412, 2025-12-04:
I apologize for the delayed response. This appears to have fixed the issue. Thank you.

closed 2025-12-04 by Razor412