prusa3d/Prusa-Firmware-Buddy #4306: [FEATURE REQUEST] allow side lights to be set for infinite duration

mogorman, 2024-11-23:
### Printer model

XL

### Describe the feature

I take timelapses as I print. the problem I am having is the light strips dim during the printing I would like to use the M151 command to set the led to bright white. the problem is i can only set it for X ms. Setting it to 0 only works with AFS

### Expected functionality

Being able to pass D0 to the M151 command

ninjubohra, 2024-11-28:
I think this capability is already available

Under the Tune Menu (I have a print currently running)
Scroll all the down to **User Interface**
And there should be **RGB Side Strip Dimming** toggle which you can turn off

[image]

mogorman, 2024-12-19:
this worked for me but i still think its an easy fix.

closed 2025-02-25 by github-actions[bot]