prusa3d/Prusa-Firmware-Buddy #4309: [BUG] Dialog for HEPA filter on XL 5T w/ enclosure does not permit bypass

bithead666, 2024-11-24:
### Printer model

XL Multi tool with enclosure

### Firmware version

6.2.0 alpha1

### Upgrades and modifications

Enclosure

### Printing from...

USB Drive

### Describe the bug

I installed the enclosure awhile back, but it has not been on the printer very long. Nevertheless, I got the attached dialog during a print. THIS IS A TREMENDOUS ERROR. Why?  Because as you can see in the video, there is no way past it!  You can't cancel out of the LCD dialog in any form or fashion! Checked with Support and they were stumped.

### How to reproduce

This software feature was not tested at all when it was implemented since the dialog is non-functional. https://youtu.be/QjKbgt6EtT4 is a movie showing the lack of response and a picture of the dialog on the LCD screen is attached here as well.
[image]

### Expected behavior

First of all, THIS DIALOG SHOULD NEVER BE PRESENTED DURING A PRINT. It should be presented when the printer is idle, or before starting a new job. But it should never interrupt a production print. It should also be dismissible.

### Files

There was no crash. Just a hanging dialog box with a road to nowhere cul-de-sac. I cannot give you the gcode since it is a paid-for file licensed privately from the owner off printables.com.

CZDanol, 2024-11-26:
Hey, thanks for reporting and sorry for the trouble. I'm pretty sure this went through multiple hands of testing. Either you've encountered some combination of circumstances that we have not forseen or the bug might have snuck in sometime later. We well look into this.

Internal ticket: BFW-6383

mare1232, 2024-12-07:
I had today the same problem, but i cant cancel printing because i dont have spare filament for a new print

bithead666, 2024-12-07:
If you yank the power cord quickly it should try to resume printing but you will have to pause it quick, set the temp, then let it continue because another bug makes the resume printing not set the correct nozzle temperature.

Prusa-Support, 2025-03-18:
Thank you again for reporting.

Please upgrade the printer firmware.
The dialog should no longer interrupt the prints now, and the commands on the screen would work as expected.

The issue can be considered solved now but please, don't hesitate to add further feedback if needed.

Michele Moramarco
Prusa Research

closed 2025-03-18 by Prusa-Support