prusa3d/Prusa-Firmware-Buddy #4312: [BUG] Wrong Thermal Runaway Detection - Causing an Emergency Shutdown

LuckyResistor, 2024-11-25:
### Printer model

Original Prusa XL 5T

### Firmware version

6.2.0-alpha2+8702

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

Changing the filament in the print preparation screen can cause a wrong thermal runaway detection, which results in an emergency shutdown.

### How to reproduce

- Reset the printer.
- Load PLA into T1
- Load PLA into T2
- Keep T3, T4, T5 empty.
- Pick T1
- Park T1 (so T1 is the last used tool)

- Load the attached Gcode to the Printer.
	- ! via Connect
	- ! as next item in the Queue

- On the screen of the printer:
	- Setup spool join for T1 and T2 for T1
	- Select “Filament"
	- The default action should be “Load PLA” for T3
	- Select “Execute all Changes”
	- Start loading the filament.

Actual behaviour:

- Nozzle T1 is heated.
- Nozzle T3 isn’t heated.
- After a few minutes, the printer stops with an emergency shutdown, because of a thermal runaway.

The gcode file is just a print using T1 and T3, I’m not sure if it plays a role at all.
I suspect, that somehow the wrong nozzle gets heated, but the printer waits for temperature on T3 and eventually things that the thermistor isn’t working. All in all, a very strange error.

### Expected behavior

- The new filament is loaded in T3
- The print job starts.

### Files

[project.zip](https://github.com/user-attachments/files/17903441/project.zip)

CZDanol, 2024-11-26:
Hello, does this happen to you consistently, or did it happen just once?

LuckyResistor, 2024-11-26:
@CZDanol It happened twice, but not with exact the same setup. But both times when I changed filament from the print preparation screen.

The issue number you mentioned, is this one.

CZDanol, 2024-11-26:
Okay, and how many times have you done the procedure? I'm trying to understand what reproducability rate we should expect.

LuckyResistor, 2024-11-26:
I would say a low reproducibility rate. I just started a new print with exact the same procedure and it worked flawlessly this time. Also, in most cases, I change the filament beforehand, and not from this menu - so, I would say with my latest successful try, its about 2/10 times since I use this printer.

zehkaiser, 2024-12-09:
> Hello, does this happen to you consistently, or did it happen just once? Might be related to BFW-5345.

I can't find a reference to this in the repo. What is BFW-5345? I've had two thermal runaway alarms while starting a print since upgrading to the alpha. The temperature graph on the web interface shows a normal heatup cycle before failure.

CZDanol, 2024-12-10:
It's an internal ticket. If I may ask, we should be releasing another release candidate soon-ish. I suspect that it will fix this issue. If not, please let us know after the new FW is released and the issue still persists.

closed 2025-02-17 by github-actions[bot]