prusa3d/Prusa-Firmware-Buddy #4316: preheat for Z-leveling

schachtelmacher, 2024-11-26:
### Printer model

MK4S

### Describe the feature

  I would like to preheat an additional menu entry. In order to level the z-axis, the MK4S needs 175° and I always preheat the bed to 85°. I would need that as a menu item so that I have the necessary temperature for leveling before I start. That saves me time - and probably others too.

### Expected functionality

another menu item, best placed at the top under preheat

CZDanol, 2024-11-26:
This seems like a duplicate of #1795.

schachtelmacher, 2024-11-27:
Not quite, but similar - I want a menu item to preheat the bed and nozzle for Z-leveling.
Ideally, you could define your own menu items. It could be a file that you can edit and then read in via the USB stick.

closed 2025-02-04 by github-actions[bot]