prusa3d/Prusa-Firmware-Buddy #4319: [FEATURE REQUEST] User configurable filament change location

Sembazuru, 2024-11-26:
### Printer model

MK4S

### Describe the feature

Currently running 6.2.0alpha2
I was surprised after installing 6.2.0a2 that the default parking location during the M600 color change moved from right-side x=max (from the user's perspective) to left-side x=minimum. Because my MK4S is in the Prusa enclosure and situated that I normally sit just to the left of the printer (there is a support column off the front-right of the enclosure where I have my printer in my basement). The old front-right filament change is a straight line from where I sit with the enclosure doors open (thank you for the 180° door swing) making it easy to reach the extruded filament during the filament change process. With the new front-left I kind of have to reach around where the door hinges are to manage the extruded filament during the filament change process. I know that I should be able to change this filament change parking position in gcode using arguments to M600, that won't work for times when I either fire off a filament change from the printer's menu, or the M600 get triggered with a detected nozzle clog.

I'd like a configurable default location for the nozzle change in the settings menu.

### Expected functionality

I haven't decided which would be better:

1. A simple drop-down with "Left (x min), Center, Right (x max)".
2. A more granular setting where you can set the x value, y value, and z-offset value.
3. An even more granular setting where all the arguments for M600 are available to set a default.

I might prefer that second option because when doing color swaps low down (in the first few layers) on a model I'd like a little more height between the extruder and the bed, and I haven't been bothered to add the M600 arguments to my gcode in PrusaSlicer...

Sembazuru, 2024-11-26:
I think this feature should apply to all printers, but might need separate issues for the non buddy firmware based printers (like the MK2 and MK3).

CZDanol, 2024-11-28:
Internal ticket: BFW-6394

CZDanol, 2025-06-25:
If we make parking positions configurable, it would make sense to not make all of them - not just one. So from our perspective, we would implement the mechanism once and then apply it on multiple positions :)

3d-gussner, 2025-07-17:
@Sembazuru We added in the Prusa-Firmware MK2.5/s and MK3/S/+ and additional parameter `S` to store the new park position (until the printer is turned off) See [RepRap Wiki M125](https://reprap.org/wiki/G-code#M125:_Pause_print) . As the 8-bit firmware has very limited resources we didn't write these "custom" values to the EEPROM and after a reset or power off the default values are back active. This could be solved by modifying the start g-code in PrusaSlicer profile so it is "re"-set each time.

Tupson444, 2025-07-17:
>... I thought the parking after printing was handled by the End G-code ...

That is how it should work, but it seems that after executing the end commands from the gcode file, the printer (MK4/Core One) always performs nozzle park command on print end (hardcoded in the firmware), therefore anything set in the slicer is useless. See #3785