prusa3d/Prusa-Firmware-Buddy #4323: Stuck Filament Detection Force Threshold Too Low

krayvis, 2024-11-27:
### Description of the bug

Not sure if this is the right place for it, but I just had repeated stuck filament detection alerts. I went through the process to unload and re-load filament multiple times. I even tried hand-feeding filament into the extruder. Every time I restarted printing,  after 10 seconds of printing the machine would halt again. I disabled the "feature" and the printer is now running fine.  

### Project file & How to reproduce

Print long enough on an XL...

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 11

### Printer model

XL - Single Extruder

wnerq, 2024-12-18:
I believe this issue could be mitigated with the enhancement I suggested in [4352](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4352)

unrepentantgeek, 2025-02-19:
Please keep this issue open.  I've had a printer repeatedly fail prints due to stuck filament detection.  Yet when I disable the detection the prints work fine.

closed 2025-05-05 by github-actions[bot]