prusa3d/Prusa-Firmware-Buddy #4325: [FEATURE REQUEST] Dynamic Print Progress Screen i.e. for  Canceled objects

ErGeeBee, 2024-12-01:
### Printer model

MK4

### Describe the feature

The Print Progress Screen now is static and shows a graphical interface of the print plate being printed. 
It would be nice if this image were to be dynamic. For example if you were to cancel an object on the plate, that the Screen would remove or grab out the object that is not being printed anymore.

danopernis, 2024-12-13:
Unfortunately, this feature is not feasible. The image displayed on the print progress screen is static; it is generated by Slicer and embedded directly into the G-code. The MCU lacks the processing power to dynamically generate these images.

While it might be possible to store multiple pre-rendered images in Slicer, the number of potential images increases exponentially as the number of objects and combinations of canceled/not-canceled objects grows. Even using an image format with support for layers wouldn’t resolve this issue. The MCU only has 256kB of RAM, which means there isn't enough memory to support a framebuffer for blending layers.

CZDanol, 2024-12-13:
I would argue this actually is feasible - we could export another bitmap from the slicer, storing what object each pixel belongs to. Then, we could apply a pseudoshader, graying out/coloring red/doing whatever to cancelled objects.

CZDanol, 2024-12-13:
I've created an internal ticket BFW-6465 for this. Let's see if the product guys find it interesting.