prusa3d/Prusa-Firmware-Buddy #4326: [FEATURE REQUEST] Forget WIFI network

Raddic22, 2024-12-02:
### Printer model

MK4, MK4s, Prusa XL, Prusa Core One

### Describe the feature

Under the network settings somewhere it would be good to be able to forget a wifi network. Currently the only way to do this is with a factory reset which also requires all the calibration to be carried out again.

### Expected functionality

Settings > Network > Wi-Fi > Forget Network

CZDanol, 2024-12-02:
Hello, we are already working on a feature like this (internal ticket BFW-5353) :) It won't make it to 6.2, but should fit into the next release.

Raddic22, 2024-12-02:
Awesome, great to hear!

CZDanol, 2025-01-10:
Hello just to let you know, the feature is almost ready, but it will likely not make it to the Core One release firmware, sorry. So, the release after that.

Raddic22, 2025-01-10:
Perfect, thanks for the update. Glad to see this feature is still progressing.

closed 2025-11-19 by github-actions[bot]