prusa3d/Prusa-Firmware-Buddy #4329: [BUG] Incorrect detection of filament and sudden crash and restart

WereCatf, 2024-12-03:
### Printer model

MK4

### Firmware version

6.1.3

### Upgrades and modifications

None

### Printing from...

USB drive

### Describe the bug

Filament ran out during print, printer asked to change to new one. The leftover filament unloaded perfectly fine but printer insisted it's still there and did not react to any change when I tried to trigger the sensor. I did verify there was no filament inside and no reason for it to think there was, so this must be a bug. This is also literally the only time I've had the printer not detect filament correctly, so that also suggests this is a bug and not an issue with the sensor.

Then when I tried to hold the scroll wheel down hoping I could force the printer to bypass that, it instead brought up the Z-axis adjustment dialog, crashed and rebooted, ruining my multi-hour print -- second bug here. Also, after the printer rebooted, it did detect that there was, indeed, no filament, so this further suggests some sort of a bug related to the reading of the sensor when unload filament mid-print.

### How to reproduce

Run out of filament mid-print and hope you run into the false detection of filament?

### Expected behavior

Printer should obviously detect correctly if filament is or isn't inserted. It also should not bring the Z-axis adjustment dialog up and subsequently crash when there is the dialog telling the user to unload filament up.

CZDanol, 2024-12-03:
Hello, you probably have debris in your filament sensor, please contact our support.

WereCatf, 2024-12-03:
@CZDanol So, you're saying that the firmware crashing and rebooting can magically remove debris from the sensor...? Or that you didn't actually read what was said and just jumped to a conclusion?

CZDanol, 2024-12-03:
I did read what you said, but not thoroughly enough.

WereCatf, 2024-12-03:
At least the crash should get fixed, ie. one shouldn't be able to bring up the Z-level adjustment dialog when there is already another dialog up.

CZDanol, 2024-12-03:
Definitely :) Internal ticket: BFW-6417

closed 2025-02-10 by github-actions[bot]