prusa3d/Prusa-Firmware-Buddy #4330: [BUG] Incorrect reset value for extruder steps per unit in experimental settings

joshuafocht, 2024-12-03:
### Printer model

MINI+

### Firmware version

6.1.2

### Upgrades and modifications

filament runout sensor

### Printing from...

USB Drive, PrusaLink, and Prusa Connect

### Describe the bug

When resetting the extruder steps per unit under experimental settings it defaults to 1 not the stock value of 325 (for a mini/+).

### How to reproduce

1. Goto experimental settings by holding click on the Hardware menu of Settings for 2-3 seconds.
2. Click reset extruder steps per unit.
3. Extruder steps per unit gets set to 1.

### Expected behavior

Clicking reset extruder steps per second should set it to 325 not 1. (For the mini/+)

### Files

https://1drv.ms/v/s!ArmBR5ZkHUhmk8tYQyPg-7nA_JrzDQ?e=joetqc

bkerler, 2024-12-08:
@joshuafocht I'm looking into this issue.

So far I couldn't see any issue that might cause this at first sight, so I will try to emulate to see what goes wrong.

Just for reference for the other devs:
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/8a788ae4e23d0fc79cf15c4c2dde05168b95b6bb/src/gui/screen_menu_experimental_settings_release.cpp#L55

Being set here:
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/8a788ae4e23d0fc79cf15c4c2dde05168b95b6bb/src/persistent_stores/store_instances/config_store/store_definition.hpp#L409

Which is hardcoded to 325 here:
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/8a788ae4e23d0fc79cf15c4c2dde05168b95b6bb/include/marlin/Configuration_MINI.h#L678

bkerler, 2024-12-08:
Most probably caused by this :
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/8a788ae4e23d0fc79cf15c4c2dde05168b95b6bb/src/persistent_stores/store_instances/config_store/old_eeprom/eeprom_v10.hpp#L92

bkerler, 2024-12-08:
@joshuafocht Two questions:
1. Do you use the release or debug firmware ?
2. Did you try to reset the eeprom and the bug still consists ?
Thanks

joshuafocht, 2024-12-09:
I use the release firmware.
I do not know how to reset the eeprom, so if you can tell me how to do that I would be happy to try.

bkerler, 2024-12-12:
@joshuafocht Using "Settings->System->Factory Reset->Reset Settings & Calibrations". But I just verified the bug happens in the simulator as well, so no need to reset the eeprom. I will try to sort out what happens now.

bkerler, 2024-12-12:
Ok, we identified the issue and I wrote a PR for it :)

joshuafocht, 2024-12-12:
Is this just a GUI issue?
Because when I tried to print it was under extruding like crazy.

bkerler, 2024-12-13:
It is a GUI issue leading that an incorrect value is written to eeprom incorrectly after reset. To be precise, the extruder is inverse, so in fact it isn't 325 but -325 in reality. As the value is below 1 and the slider goes from 1 to 999 (I think), the value is being set to 1.
Until the firmware is fixed, make sure to set it to 325 after resetting.

joshuafocht, 2024-12-17:
Ok, thanks for your help!

CZDanol, 2025-01-30:
Hello, 6.2 has been released: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0

The issue should be addressed, so closing. Have a nice day! :)

closed 2025-01-30 by CZDanol