prusa3d/Prusa-Firmware-Buddy #4331: [BUG] Missing serial message when printer stop before job end completly

samdurnxn, 2024-12-04:
### Printer model

MK4S with MMU3

### Firmware version

6.2.0.alpha2

### Printing from...

PrusaLink

### Describe the bug

Hello, before the firmware 6.2.x alpha, when user stop the print for any reason, a message on serial interface was sent.
But in 6.2, the feature was removed and any message is sent.

I found in the previous firmware the function which was executed but the file was removed in the commit  f73d689ab6498a2a9c6b4fb5d1130b891db2c788 related to the ticket BFW-5653

The function : 
```
void media_print_stop(void) {
    if ((media_print_state == media_print_state_PRINTING) || (media_print_state == media_print_state_PAUSED)) {
        close_file();

        // Do not remove, needed for 3rd party tools such as octoprint to get status that the gcode file printing has finished
        SERIAL_ECHOLNPGM(MSG_FILE_PRINTED);

        media_print_state = media_print_state_NONE;
        queue.sdpos = GCodeQueue::SDPOS_INVALID;
    }
}
```
Is is possible to restore this function  or at least add the line which sent the message on serial in the new stop function ? I think everyone which use a 3d party tools have this issue that they can't know when printer stop the current job.

Thanks. 

### How to reproduce

- Connect to printer with serial cable to show every messages
- Begin any job
- Stop the job with interface
- No message will be sent on serial to indicate that printing file is done

### Expected behavior

In the previous firmware, the printer sent the message "Done printing File" on serial when job was stop

CZDanol, 2024-12-04:
Hello, thanks for the report. Internal ticket: BFW-6420

samdurnxn, 2025-01-09:
thanks for the update .

CZDanol, 2025-01-30:
Hello, 6.2 has been released: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0

The issue should be addressed, so closing. Have a nice day! :)

closed 2025-01-30 by CZDanol

samdurnxn, 2025-01-31:
Thanks ! It's working correctly.