prusa3d/Prusa-Firmware-Buddy #4332: [BUG] 6.2.0-alpha2+8702 - purge occuring in motion back to print when resuming after end of filament detected

JGeiger2024, 2024-12-04:
### Printer model

MK4 + MMU3

### Firmware version

6.2.0-alpha2+8702

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

I was running a 2-color print (extruder 3 and 4) when the filament broke inside the tube between the MMU3 and the spool.  When the end of the filament was detected at the MMU3, the print was interrupted, the nozzle was moved to the left front of the build plate, an amount of filment was purged and then I was prompted to address the issue.  The MMU3 shuttle had moved all the way to the right and a short length of the filament was poking up out of the MMU3. I pulled out the remaining piece between the MMU3 and the extruder and manually fed the filament back up near to the entrance of the MMU3.  I followed prompts to recover and manually pushed the filament far enough to be engaged by the MMU3.  The filament was successfully fed to the extruder and an amount of filament was purged.  I followed the prompt to acknowledge that the correct color was present.  On the move back to resume the print, about 25mm more material came out of the nozzle.  I quickly picked it off of the nozzle because if I hadn't it would have ended up being deposited on my print.

### How to reproduce

I would expect this to be reproduceable but have not tried yet

### Expected behavior

If there were any more material that needed to be purged, it should have been completed before actually resuming the print.

CZDanol, 2024-12-05:
Hello, I would say that this is rather an intended feature. When the nozzle is heated, the filament is constantly oozing a bit. If we didn't purge a bit just before starting the print, all the filament from the nozzle could be oozed out and the printer would not be extruding anything for a bit after resuming.

However it would probably be a good idea for us to advertise that there will be some extra extrusion after pressing the button.

murk-sy, 2024-12-07:
It's a pain point for me too, though without MMU3, mostly on the Mini. Part of the problem is that after the extrusion, the hotend moves slowly to the correct position, which causes even more ooze/extrusion. Though at this point I'm no longer sure if it's actually ooze or just a consequence of issue 2997.

closed 2025-02-13 by github-actions[bot]