prusa3d/Prusa-Firmware-Buddy #4336: [BUG] Issue to open wifi wizard during print

xXNoceboXx, 2024-12-09:
### Printer model

MK4

### Firmware version

6.1.3+7898

### Upgrades and modifications

None

### Printing from...

USB Drive

### Describe the bug

During a print, I decide to connect to wifi.
For that I click to open wifi wizard 2 times and have an issue.

### How to reproduce

During a print, I decide to connect to wifi.
For that I click to open wifi wizard 2 times and have an issue.

### Files

[image]

https://github.com/user-attachments/assets/5e49244d-ca25-41af-bbb9-aa4c21d0c462

CZDanol, 2024-12-09:
Hey, thanks for the report. Internal ticket: BFW-6443

CZDanol, 2025-01-30:
Hello, 6.2 has been released: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0

The issue was somewhat addressed - it should no longer result in a redscreen. It is still not advisable to open the wi-fi wizard during print as it is a gcode and executing it will pause your print.