prusa3d/Prusa-Firmware-Buddy #4337: [BUG] Prusa XL homing fail CRASH

Investegate, 2024-12-09:
### Printer model

Prusa XL 5TH

### Firmware version

6.2.0 alpha2

### Printing from...

USB

### Describe the bug

Doing a cold pull using the integrated function in the control tab will result in a crash if you leave the printer at rest and the steppers switch off. The printer then moves all the way to the right, which causes the tool head to unlock and crash.
Normally, the printer should go back home after deactivating the steppers. To do this, it moves to the front left. But it should never move to the right.

Video: https://www.dropbox.com/scl/fi/ibqe7p6rbnno1wapdnhra/IMG_1052.mov?rlkey=kjrjcmg8ym5az4z01e2s9n8pd&st=uucof9zm&dl=0

### How to reproduce

Do a cold pull with the integrated function in the control tab. Let the printer do its work and then leave it alone until the steppers deactivate automatically. No matter what you do to the printer, it first moves to the right and unfortunately unlocks the head and crashes.

### Expected behavior

The printer should never move to the right stop if it has to re-home after automatically deactivating steppers.

CZDanol, 2024-12-09:
Hello, thank you for your report. This looks like BFW-6259, which we've already investigated and the fix should go out in the next release candidate. Please let me know if the issue persists after the next 6.2.0 candidate is out.

CloudStrife1983, 2024-12-31:
Same problem here. 2 times in a row. During homing, the tool drives to much to the right and the tool was unlocked.

Prusa-Support, 2025-06-03:
The issue was potentially solved by the first available public final release of late January (6.2.0).
Feedback would be appreciated to close the issue now, or before it falls stale.

Michele Moramarco
Prusa Research

Investegate, 2025-06-03:
> The issue was potentially solved by the first available public final release of late January (6.2.0). Feedback would be appreciated to close the issue now, or before it falls stale.
> 
> Michele Moramarco Prusa Research

Yes it is fixed. But i had to do a hard reset.

closed 2025-06-03 by Investegate