prusa3d/Prusa-Firmware-Buddy #4338: [FEATURE REQUEST] stealth modes status via gcode

mogorman, 2024-12-09:
### Printer model

prusa xl

### Describe the feature

I would like to be able to query the current status of the stealthmode of the printer. like the other prusa firmware.
like described here https://github.com/prusa3d/Prusa-Firmware/pull/4388

### Expected functionality

`M915
M915 Q
ok
Mode: Silent
ok

M914
M914 Q
ok
Mode: Normal
ok

<Normal on display>
M914 Q
Mode: Normal
ok

<Change to Stealth on display>
M914 Q
Mode: Silent
ok`

CZDanol, 2024-12-17:
Okay, so apparently there is already M1940 for disabling the stealth mode and M9150 for enabling it :)

CZDanol, 2024-12-17:
Oh, we're missing a functionality to query it - I guess we will add it to one of these.