prusa3d/Prusa-Firmware-Buddy #4339: [BUG] Extruder skipping steps when unloading filament

rangido, 2024-12-09:
### Printer model

Prusa MK4S

### Firmware version

6.1.3

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

When unloading the filament on the MK4S, the extruder is "skipping" during the "ramming" operation before unloading the filament. This issue has started since I upgraded from the MK4 to the MK4S using the upgrade kit.

I suspect that the CHT Nextruder Nozzle does not tolerate the ramming as well as the conventional nozzle (tries to force the filament through 3 small paths to the tip versus the single path of the conventional nozzle). This causes "skipping" of the extruder during the "ramming" process. 

Except for this "skipping" during unloading, my printer has been working flawlessly. It has been producing good prints and there hasn't been any "skipping" during printing.
I have seen a few people with the same issue I am experiencing. All of them seem to have a CHT nozzle installed, which leads me further to believe that this is the issue.

The used filament type is PLA.

### How to reproduce

Unload the filament using the menu option "Unload Filament" while using a CHT nozzle.

### Expected behavior

The filament should be unloaded without the extruder skipping steps.

### Files

https://github.com/user-attachments/assets/f84f859a-e48e-4cf1-a46c-d2d78517cebd

(The little blue dot on the filament is a marker to better visualize the movement of the filament)

me25542, 2024-12-09:
I have also had this issue on my MK4, using a 0.4mm Bondtech CHT with v6 adapter. A setting like "HF nozzle" would be nice, so the firmware can have an optimal unload routine for both types of nozzles.

CZDanol, 2024-12-10:
Hello, there indeed is a setting for the nozzle type in Settings → Hardware :)

CZDanol, 2024-12-10:
Hmm, maybe it's new in the 6.2.0-alpha, sorry, maybe I lost track a bit about the release versions

rangido, 2024-12-10:
> Hmm, maybe it's new in the 6.2.0-alpha, sorry, maybe I lost track a bit about the release versions

I have flashed the new 6.2.0-alpha 2 and turned the option „Nozzle High-flow“ on. My issue still persists. The extruder is still skipping during the „ramming“ operation when unloading the filament.

Bazimek, 2024-12-10:
Same issue for my MK4S.

panneff, 2024-12-13:
Same issue here on my MK4 -> MK4S printer, on the latest alpha build.

alphadecay, 2025-01-04:
Yet another instance of this with my MK4S. It doesn't prevent unloading, but it does start to skip/grind the filament during the ramming.

CZDanol, 2025-01-06:
Hello, we will deploy a modified ramming sequence for HF-nozzles. Internal ticket: BFW-6540.